svenhb/GRBL-Plotter #93: How to move the dxf trajectory display

lianzaozi, 2019-12-09:
In addition to zooming, how to move the display of dxf track? Pressing the mouse wheel like AutoCAD software can move the drawing area. Similar functions

svenhb, 2019-12-09:
If you press middle mouse button on selected (yellow) path, you can move it...
After move in 2D-View, apply transformation via right-click menu

lianzaozi, 2019-12-09:
[image]
No "Apply Conversion" option via right-click menu

lianzaozi, 2019-12-09:
The function of pressing the middle mouse button to move the mouse to control the track display area can be compared with libreCAD software. After you draw a lot of graphics in the drawing area in the libreCAD software, and then press the middle mouse button to move the mouse, you will find that the entire drawing in the drawing area moves with your mouse. What I want to express is that. If my expression is not very clear, you can draw a dozen straight lines in the libreCAD software, and then move the mouse while holding down the middle mouse button, and you will observe the scenario I described, which is what I want The meaning expressed. At present, the plotter software is zooming in on the track display area, and there is no way to move this area while keeping the zoomed state unchanged.

svenhb, 2019-12-09:
"Move selected path" will apply the offset to the GCode...
Perhaps better naming needed?

lianzaozi, 2019-12-09:
I hope I have accurately expressed my thoughts

svenhb, 2019-12-09:
Now I got it.

svenhb, 2020-01-07:
With latest release 1.3.3.0
Middle mouse button will move complete view or selected form

closed 2020-01-07 by svenhb