svenhb/GRBL-Plotter #94: Idioma

Wdespontin, 2019-12-12:
Good afternoon, congratulations on always improving this software.
A version of GRBL Ploter 1325 that was recently released with Portuguese translation.
Some words are going beyond the windows where you can enter numbers by following red underlined photos of what happens.
English - Brazil is a bit confusing in some words.
It's like translating Portuguese into English and vice versa, there are some wrong interpretations of the words.
But there must be an error in the translator, also on the first graphic screen marked with blue ray, where the white zoom screen appears, which allows much to follow or follow the trace of the drawing that could be left in another fixed location, thanks

svenhb, 2019-12-12:
Yes I know .... I used google translator and was hoping for correction by users - like you.
Give me some help to shorten words / find abbreviations.

Wdespontin, 2019-12-13:
Good morning, I can change or delete this text here and send it to you,
edit it, this is an open program, how do I have any editable file that I
can fix and send you, thanks

*Att; *
*Valter*

svenhb, 2019-12-13:
How to edit language files:
a) Easy method for me: you download all files and use a tool like "ResxTranslator" (https://github.com/HakanL/resxtranslator).
b) Easy method for you: I setup a excel or CSV file you could edit with any program.

Finally send me the files. Tell me your choice, for a) I need to update files on github first...

Wdespontin, 2019-12-13:
Good morning, I already downloaded the resx translator, you can send me the
CSV file or I can find it in the GRBL Ploter folder, thanks

*Att; *
*Valter*

svenhb, 2019-12-13:
Hi, a third option is:
making screenshot, and mark the changings as done here: #78 

[setup.zip](https://github.com/svenhb/GRBL-Plotter/files/3960789/setup.zip)
[mainForm.zip](https://github.com/svenhb/GRBL-Plotter/files/3960790/mainForm.zip)

svenhb, 2019-12-13:
If you download setup or mainForm file, please check for actual version 1.3.2.5a

svenhb, 2020-01-07:
Up to now no updates for language corrections...

closed 2020-01-07 by svenhb