svenhb/GRBL-Plotter #95: FastColoredTextBox display Chinese garbled solution

lianzaozi, 2019-12-20:
FastColoredTextBox displays Chinese garbled characters, the problem of overlap has been resolved, please use the following library https://github.com/ks3dev/FastColoredTextBox
At the same time, when storing the Gcode code, you can choose UTF-16 BE or UTF-16 LE as the text format. If you use UTF-8 and ANSI storage, the Chinese comments in the G code will not display properly after opening.

svenhb, 2019-12-20:
Thank you, I will try soon

svenhb, 2019-12-20:
Check new release

lianzaozi, 2019-12-21:
Thank you very much, the problem has been solved, and I wish you a Merry Christmas in advance.

lianzaozi, 2019-12-21:
Thank you very much, the problem has been solved.

svenhb, 2020-01-07:
Problem solved with suggested change

closed 2020-01-07 by svenhb