svenhb/GRBL-Plotter #97: where is set new option "Don't change pen width on zooming"？

lianzaozi, 2020-01-08:
This software is already great, but I can't find "Don't change pen width on zooming"

lianzaozi, 2020-01-08:
Hi, svenhb, thank you for providing such a good platform for lovers to communicate. Where can I buy a hardware control board for use with GRBL-Plotter? If you have a purchase link, please send it to me. I mainly use it for laser engraving. It is best to have a control board with PWM or analog to control the laser energy. I am currently in China. Thank you.

svenhb, 2020-01-08:
Hi, for "Don't change pen width on zooming" I added a checkbox in [Setup - Joystick/Colors].
I bought all my boards on ebay, just search for "Arduino CNC". For 10€ you sometimes get the Arduino, the CNC-Shield and motor drivers...

lianzaozi, 2020-01-08:
Thank you for your reply.

svenhb, 2020-01-13:
Probably found

closed 2020-01-13 by svenhb