svenhb/GRBL-Plotter #98: The mouse clicks on the primitives in the three blue boxes. The software selects the elliptical arcs in the black boxes.

lianzaozi, 2020-01-08:
The mouse clicks on the primitives in the three blue boxes. The software selects the elliptical arcs in the black boxes.

test.zip file is the dxf file for my test

[image]

[test.zip](https://github.com/svenhb/GRBL-Plotter/files/4034382/test.zip)

lianzaozi, 2020-01-08:
[image]

lianzaozi, 2020-01-08:
[Software error screenshot.zip](https://github.com/svenhb/GRBL-Plotter/files/4034391/Software.error.screenshot.zip)

svenhb, 2020-01-08:
The Problem seems to be the import of DXF 'Point' geometry - without points it works...
I will fix it.

lianzaozi, 2020-01-08:
Thank you for your reply.

It is best to add a mouse to select geometric primitives. All the selected primitives are selected, and then these selected primitives can be moved as a whole. Box selection is the same as LibreCAD software. Then right-click and add an item in the pop-up menu to set the distance of the x direction and the y direction of the movement. Set the distance. The selected geometrical elements just moved to the new position with the set value

lianzaozi, 2020-01-08:
Or it also supports ctrl + mouse click multiple selection

commit 7a2915d0 (2020-01-13): Version 1.3.3.2

#102 should be fixed (missing 'close' for polyline)
'Reload' button position changed
#101 translation changed
#100 fixed
#98 fixed (DXF points wrong conversion)
closed 2020-01-13 by svenhb