svenhb/GRBL-Plotter #100: A very strange thing happened --- the software automatically increased the straight line

lianzaozi, 2020-01-09:
A very strange thing happened. The rectangle of the dxf file I tested was overlapped by three rectangles of the same size (as shown in Figure 1). The semicircle was also overlapped by three geometric figures of the same size Together, this dxf was opened for the first time in the plotter software and displayed as shown in Figure 2. However, when opened again with the plotter software, it was displayed as shown in Figure 3. (The blue line in Figure 3 is The software itself shows that this line does not appear in the dxf file. However, this line was not displayed when it was first opened with the plotter software, and it was displayed the second time, which is strange)
[image]
[image]

[image]
[testdxf.zip](https://github.com/svenhb/GRBL-Plotter/files/4039150/testdxf.zip)

svenhb, 2020-01-09:
Which 'Use case' did you select? Perhaps you can provide the generated G-Code?

lianzaozi, 2020-01-10:
[image]
[test.zip](https://github.com/svenhb/GRBL-Plotter/files/4043407/test.zip)
[test_dxf.zip](https://github.com/svenhb/GRBL-Plotter/files/4043409/test_dxf.zip)

svenhb, 2020-01-10:
Try https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1331_publish.zip
Should be fixed:
Language in menu points
Additional Line
Hopefully "Reload SVG" button now visible

lianzaozi, 2020-01-11:
ok thanks

svenhb, 2020-01-13:
Should be fixed in 1.3.3.2 https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1332_publish.zip
#102 should be fixed (missing 'close' for polyline)
'Reload' button position changed
#101 translation changed
#100 fixed
#98 fixed (DXF points wrong conversion)

closed 2020-01-13 by svenhb