svenhb/GRBL-Plotter #99: Summary of some issues that need improvement

lianzaozi, 2020-01-09:
Summary of some issues that need improvement:

1. In the simulation, not only the long straight line (corresponding to G01), the long-air travel distance (corresponding to G00), the animation flashed, but also the entire circle corresponding to G02 and G03 flashed.
2. "Machine control" => "coordinate System", the information in this form is not translated into Chinese
3. "Machine control" => "DIY Control", the information in this form is not translated into Chinese
4. "Machine control" => "Control streaming", the information in this form is not translated into Chinese
5. "Machine control" => "Control 2nd GRBL", the information in this form is not translated into Chinese
6. For the problem that the image is not centered on the mouse pointer when zoomed in, you can solve this problem by adding a zoom-in / zoom-out menu to the right-click pop-up menu. After selecting the mouse, you can zoom in on the part selected by the mouse frame. And the center display, zoom out is also the case so that it will not cause the problem of not zooming at the current pointer point of the mouse (refer to the practice of CNCInfusion software, the link is https://github.com/nm156/CNCInfusion) (OpenCNCPilot this software The zoom is also better, the link is https://github.com/martin2250/OpenCNCPilot, the software operation steps are: mouse wheel: zoom in / out; middle mouse button: press the middle mouse button to move the mouse to move the screen (Right mouse button: hold down the right mouse button and move the mouse to rotate the view; double-click the middle mouse button: the enlarged or reduced view can be adaptively displayed in the window)

[image]

7. About the setting item of subdividing the curve into small line segments. The test result on my computer is that no matter how the value is set, the software does not change the number of subdividing the curve into small line segments (this should not be completely resolved after the last correction. Question, I hope you test it on your computer)

[image]

8. The subdivision of the arc into small line segments does not have any effect. No matter how the parameters are changed, the number of small segments into which the arc is subdivided will not change. (This problem should not be completely solved after the last correction. , I hope you test it on your computer), and when the "Avoid G2 / G3 commands" option is not checked, the G code generated by the arc is represented by G01, the correct way is not to uncheck "Avoid G2 / G3 commands "option arc to generate G02 and G03, check this software to divide it into small straight lines according to the value in the textBox?
[image]

lianzaozi, 2020-01-09:
This rectangle is superimposed by three rectangles of the same size. This graphic is imported into the software. The software does not check for any geometric primitive overlap. As a result, the rectangular path is processed three times. Sometimes it is common for two identical geometric primitives to overlap due to drawing reasons. Therefore, it is better for the software to have the function of setting whether to automatically delete overlapping geometric primitives. When there are overlapping primitives, when the dxf file is imported into the software, the overlapping primitives should be highlighted in the display area, and a pop-up window will tell the user. There are overlapping primitives in this dxf file.

[image]
[test1.zip](https://github.com/svenhb/GRBL-Plotter/files/4038450/test1.zip)

svenhb, 2020-01-09:
About: 7 / 8 it's working for me:
[image]
   
  
Don't forget to reload the file after changing any setting  
[image]

[Test_arc.zip](https://github.com/svenhb/GRBL-Plotter/files/4041707/Test_arc.zip)

lianzaozi, 2020-01-10:
No matter how I set the segmentation parameters, the number of Gcodes generated does not change at all.

[22222.zip](https://github.com/svenhb/GRBL-Plotter/files/4043431/22222.zip)
[test_dxf.zip](https://github.com/svenhb/GRBL-Plotter/files/4043436/test_dxf.zip)

lianzaozi, 2020-01-10:
[image]
No matter how I set the segmentation parameters, the result is the same

lianzaozi, 2020-01-10:
[image]
I downloaded these two documents, both of which have been tested without button and checkbox
[image]

lianzaozi, 2020-01-10:
But using vs2019 to directly open the source code and recompile will not have buttons and checkboxes, you must operate as follows:
You need to click the setting form code in the solution explorer, display the setting interface form, and then place your mouse in the lower right corner of the form. When the cursor becomes a double-headed arrow 45 ° obliquely downward, drag to the lower right Form, and then select the "Read SVG, DXF, Drill file" button with the mouse (that is, move the cursor to the control and left-click once) and select the "Show use case selection on graphics import" check box (that is, move the cursor to Left-click on the control once), select "Generate" => "Generate Solution" in the menu bar and regenerate all once to solve the problem of no button and checkbox.

lianzaozi, 2020-01-10:
It should be that the size of the window is too small to cover up. Make the window slightly larger and re-debug

svenhb, 2020-01-10:
>I downloaded these two documents  

This means you used the exe file from "GRBL-Plotter_1330_publish.zip"? And the buttons are missing?
Strange, how could I check this? If I switch GRBL-Plotter to chinese language the buttons are still there.

lianzaozi, 2020-01-10:
I also don't quite understand why the button and checkbox are recompiled after dragging the form, and I'm still looking for the specific reason.
The 55555 file is a .gif file. This file shows my entire operation. I just opened the settings window and dragged the following form.

[image]

If you can, you can use the TeamView software to remotely control my computer. If you have a Microsoft Skype account, you can send your account to my mailbox, so that you can remotely control my computer and send me a message via Skype software.

svenhb, 2020-01-10:
I will try to set these buttons on top (z-order) recompile and upload a 1.3.3.1_publish.zip - perhaps in 5 hours from now.

lianzaozi, 2020-01-10:
Regarding the problem that the setting parameters of the curve and ellipse subdivision into small line segments cannot take effect, look at the specific reason. My operation process is in the file 22222.zip. Download this file and decompress it to get a 22222.gif. This gif file can display During my entire operation, the two parameters "Avoid G2 / G3 commands Step width on circumferece" and "Number of line segments in curve" no matter how I set them, they have not brought any changes.

svenhb, 2020-01-10:
You need to press "reload", don't select again a "Use-case" - this will overwrite any previous changes in setup. In "Use-case" window use the yellow button, to go on without changing...
That's the idea of "Use-Case" to reset any "strange" settings for import...

lianzaozi, 2020-01-10:
Thanks, I tested it and it worked very well. You're great

svenhb, 2020-01-10:
Sorry for the confusion, I started a FAQ here: https://github.com/svenhb/GRBL-Plotter/wiki#grbl-plotter

lianzaozi, 2020-01-11:
ok

svenhb, 2020-01-23:
Problem is already solved

closed 2020-01-23 by svenhb