svenhb/GRBL-Plotter #101: Tradução Inicial

Wdespontin, 2020-01-12:
[image]
[image]
[image]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

svenhb, 2020-01-12:
To have your text as real text (not as a screenshot) would help to make the changes (copy & paste).
Do you think the original grbl messages should be translated also (3 IDLE - Esperando)?

Wdespontin, 2020-01-12:
Follows the changes, from portuguese to english translator very erroneous
or vice versa

*Att; *
*Valter*

svenhb, 2020-01-12:
Thanks,
these are the other grbl specific messages...

Alarm	
Check	
Door	
Hold	
Home	
Idle	         Ocioso
Jog	
Probe	
Run		Ecexutando
Sleep

Wdespontin, 2020-01-12:
[image]
[Comando.D.txt](https://github.com/svenhb/GRBL-Plotter/files/4051587/Comando.D.txt)

svenhb, 2020-01-13:
Check update for translation
Should be fixed in 1.3.3.2 https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1332_publish.zip
#102 should be fixed (missing 'close' for polyline)
'Reload' button position changed
#101 translation changed
#100 fixed
#98 fixed (DXF points wrong conversion)

svenhb, 2020-01-23:
Thank you for your help.

closed 2020-01-23 by svenhb