svenhb/GRBL-Plotter #102: 绘图仪软件会把dxf文件中的某些几何图元丢掉

lianzaozi, 2020-01-13:
Plotter software will discard some geometric primitives in the dxf file (these geometric primitives not displayed by the plotter software should not be parsed)

Open with plotter software
[image]

Open the same dxf with CAD software
[image]

lianzaozi, 2020-01-13:
This is the dxf file I used for testing
[WenZiGuiJi_Test.zip](https://github.com/svenhb/GRBL-Plotter/files/4051828/WenZiGuiJi_Test.zip)

lianzaozi, 2020-01-13:
I am curious, the function of generating GCode with text in the plotter software can not achieve the effect of this hollow text.

svenhb, 2020-01-13:
Did you create the text with LibreCad? Because in LibreCad the file "WenZiGuiJi_Test" just contains polyline, not text.
In grbl-Plotter I copied all fonts, which are available in LibreCad...

lianzaozi, 2020-01-13:
[image]
No effect

svenhb, 2020-01-13:
[image]

svenhb, 2020-01-13:
Should look like this:
[image]

svenhb, 2020-01-13:
Should be fixed in 1.3.3.2 https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1332_publish.zip
#102 should be fixed (missing 'close' for polyline)
'Reload' button position changed
#101 translation changed
#100 fixed
#98 fixed (DXF points wrong conversion)

lianzaozi, 2020-01-14:
[image]
Thank you, the effect is great

lianzaozi, 2020-01-14:
Which font of the plotter software can I choose to make hollow Chinese characters? I have tried all the fonts provided in the plotter software and can only produce single-line Chinese characters. No font can produce hollow Chinese characters

svenhb, 2020-01-14:
To use (probably) any font you like, you write your text in inkscape, select contour and fill, then convert object to path, then copy and paste to grbl-Plotter.    
https://inkscape.org/de/doc/tutorials/advanced/tutorial-advanced.html   
  
Perhaps here are matching fonts? https://www.dafont.com/theme.php?cat=201

svenhb, 2020-01-23:
Already fixed

closed 2020-01-23 by svenhb