svenhb/GRBL-Plotter #103: Plotter software reads dxf files containing text and svg files, an error occurs

lianzaozi, 2020-01-14:
Plotter software reads dxf files containing text and svg files, an error occurs

Plotter software reads screenshots of dxf files with text
[image]

CAD software reads a screenshot of the same DXF file with text
[image]

Plotter software reads screenshots of svg files with text
[image]

Screenshot of Inkscape software reading the same SVG file with text
[image]

svenhb, 2020-01-14:
grbl-Plotter can't read real text from SVG - too complicated to implement.
Workarround for SVG: convert text to path, then it works.

lianzaozi, 2020-01-15:
But the text in the dxf file cannot be parsed normally.

svenhb, 2020-01-15:
I already use the DXF font (*.lff) when "Generating G-Code from Text". So I can support Text-entity from DXF file. Inside DXF file, the text, font-family, size, position etc. are stored...

lianzaozi, 2020-01-15:
Thank you very much.
I tested it with LibreCAD to create a dxf file with text. The plotter software does recognize the text in a dxf file with text created with LibreCAD. The fonts in LibreCAD software do not support Chinese. I tested all the fonts in LibreCAD and displayed garbled characters when typing Chinese.

But read the text created by AUTOCAD software with dxf plotter software that can not recognize the text.

svenhb, 2020-01-15:
In Inkscape you can use any windows-font (I think), I tried with chinese text and at least there are two fonts showing difference style for chinese text.
[image]

lianzaozi, 2020-01-15:
Thank you very much for your answer
Inkscape supports Chinese very well, LibreCAD software does not support Chinese. Now there is only the method of converting text to track and saving as SVG in Inkscape.

svenhb, 2020-01-15:
* You can copy the path in Inkscape and paste in GRBL-Plotter (on 2D view)  
* You can save as SVG and load in GRBL-Plotter  
* You can export to DXF in Inkscape and go on in LibeCAD - still as path, not as text  
  
You have a lot of options

lianzaozi, 2020-01-15:
ok

svenhb, 2020-01-23:
As described, it's too complicated to implement real text import from SVG.

closed 2020-01-23 by svenhb