svenhb/GRBL-Plotter #105: Regarding the authenticity of the simulation process of enhancing the long straight line (G01), long flight (G00), and full circle (G02 / G03) in the future.

lianzaozi, 2020-01-15:
Regarding the authenticity of the simulation process of enhancing the long straight line (G01), long flight (G00), and full circle (G02 / G03) in the future.

I think that you can take long straight lines, go through the air, and complete circles to subdivide them into small segments, and then save the subdivided lines into a generic set. When the simulation starts, the position of the points can be redrawn every other timing period based on the data of the small line segments taken from the generic set. When the G code is actually generated, the full circle still generates the GCode corresponding to G02 / G03.

lianzaozi, 2020-01-15:
The effect is similar to this, but I do n’t do too much. If you have a better idea, you can code it according to your own ideas.
[image]

svenhb, 2020-01-15:
I already implemented intermediate steps for G0 / G1.
For G2 / G3 comes next.

lianzaozi, 2020-01-16:
Ok,
thank you for your reply

lianzaozi, 2020-01-16:
When the path is simulated, can the path of the subroutine be simulated normally, such as the following example
[image]

Can the GCode generated from pictures and GCode generated from text also be simulated normally?

svenhb, 2020-01-19:
Check new version https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1333_publish.zip
Simulation for G0,1,2,3 implemented,
also subroutines considered

lianzaozi, 2020-01-21:
Your job is great and it works well.
However, when simulating a subrouti
Your job is great and it works well.
However, when simulating a subroutine, it will not return to the starting point after the simulation is completed.

The dxf file simulates G00 G01 G02 G03, the effect is great.ne, it will not return to the starting point after the simulation is completed.

The dxf file simulates G00 G01 G02 G03, the effect is great.

lianzaozi, 2020-01-21:
[image]

svenhb, 2020-01-21:
Thanks,
I thought it's a good idea to stop at M30 with the subroutine. But I don't stop at the end at 'normal' G-Code...
Whats the better solution? Both jumps to line 1 or both stop at the end?

lianzaozi, 2020-01-22:
Glad to hear from you.
What is the effect of staying at M30 after the subroutine simulation is completed? It is common practice to return to the first line after the simulation is complete.

lianzaozi, 2020-01-22:
I know that it is better to set the speed of the idle track and the processing track separately during the simulation. Generally, the speed of the empty track and the processing track are not the same.

lianzaozi, 2020-01-22:
When sending a strange phenomenon simulation, it always stays at the beginning of each primitive for a long time before moving.

Please see my .gif below

svenhb, 2020-01-22:
I use different speeds for G0 (fix) and G1,2,3 (given feedrate).
>It is common practice to return to the first line after the simulation is complete.
Ok

svenhb, 2020-01-23:
Now it jumps back to line 0
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1334_publish.zip

closed 2020-01-23 by svenhb