svenhb/GRBL-Plotter #106: Zoom

lianzaozi, 2020-01-26:
Thanks for your work
Version 1.3.3.4 does not scale well

lianzaozi, 2020-03-18:
I have two suggestions for improvements regarding scaling:
1. Regardless of moving the trajectory to any position in the display area, when you double-click the mouse wheel quickly, the trajectory is immediately and adaptively displayed in the center of the display area (this can refer to libreCAD software).
[image]

2. Regarding zooming, the zoom magnification should not be increased by 2 times each time the mouse wheel is scrolled.

svenhb, 2020-03-22:
* 2D-View: fix strange zomm-in, -out behavior  
Reset zooming on double-click middle mouse button    

https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.3.4.3

I could fix the problem finally :-)

closed 2020-03-22 by svenhb