svenhb/GRBL-Plotter #109: Support conveyorized paper roll ?

raercan, 2020-02-04:
https://youtu.be/eO6vynBqP78

Plotters can use paper roll.

svenhb, 2020-02-04:
I don't know what to implement.
I would just add some footer-gcode to control axis 'B' to turn the paper-roll after finishing the cut.  
  You could also use as 2nd grbl-controller to control the paper-roll: https://github.com/svenhb/GRBL-Plotter/wiki#how-to-control-a-2nd-controller-with-grbl

raercan, 2020-02-04:
Actually need huge draw area. Maybe u can slice x or y side . When completed bite again draw another bite. Ploter can draw part by part . Picture by picture. Bite by bite. ( Sorry my english ) thank you.

lianzaozi, 2020-04-23:
In order to express my thoughts, I drew a schematic diagram to express, the following is a text description.
If there is a 90 * 155 picture to be drawn, but the maximum working range of the machine is 45 * 48, after the file to be processed is imported into the software, the software automatically divides the motion area to be processed according to the size of the picture file and the set machine processing range. 6 copies, and then generate the corresponding 6 copies of GCode (I use the red dotted line in the figure to simulate the six copies of the software division). If the area 1 is processed first, the machine fixture axis automatically moves by +45 in the X direction after processing (the user without a fixture axis moves manually), and then the area 2 is processed. After the area 2 is processed, the machine fixture axis automatically moves in the Y direction by -48 (users without a fixture axis move manually) and then process area 3. Then move in X direction -45 machining area 4 ...
I feel that this way I can solve large-format drawing work with a small working area. If there is a better solution, I like to communicate and solve the problem together.

[image]

raercan, 2020-04-23:
https://www.youtube.com/watch?v=jMF-TktGB-8

i was imagined like this machine.

svenhb, 2020-05-15:
Clipping is the 'magic' process to cut the graphic-elements. 
Clipping is only simple for line segments, so I will convert all circles and arcs into lines, before generating the partial graphics - let's say 'tiles'.

svenhb, 2020-07-23:
Check the new [tiling option](https://github.com/svenhb/GRBL-Plotter/wiki/Import-options#clipping) in the new release

lianzaozi, 2020-07-27:
There is a problem with the Clipping function. Below is the dxf file used for my setup and testing and the screenshot of the error.

setup
[image]

error form
[image]

test dxf file
[test.zip](https://github.com/svenhb/GRBL-Plotter/files/4980722/test.zip)

lianzaozi, 2020-07-27:
I know the cause of the error. The maximum area occupied by the geometric primitives in dxf is 297*210. When I divide it, I divide it at a distance of 100 in the X direction and 71 in the Y direction. Because 100*3>297, 71*3>210. So the software went wrong.

lianzaozi, 2020-07-27:
The file I used for testing is test.zip, which has been uploaded in the comments above.

After I set the split spacing correctly, the software now works.
But there will be straight lines in the blue area in the screenshot (image.png). I don't know what these straight lines represent.
[image]
[image]

lianzaozi, 2020-07-27:
I don't understand why the software will move to P0, P1, P2, P3, P4, P5 these positions? The gif animation shows the process of software simulation. I selected these positions with blue boxes in the screenshot (image10.png).

[image]
[image]

svenhb, 2020-07-27:
It's because of "Gcode before tile". Some how you need to move your paper to draw the next tile. I should have use other axis names...

svenhb, 2020-07-27:
My idea was to use a second Grbl controller to move a roll of paper (a small one from a cash register) for a real example, but until now I haven't had time to build the mechanics.
https://github.com/svenhb/GRBL-Plotter/wiki#how-to-control-a-2nd-controller-with-grbl

svenhb, 2020-07-28:
Check new release for fixed cut lines

lianzaozi, 2020-07-29:
The software cutting function of version V1.5.0.2 cannot be used. After the cutting function is enabled, an error will be reported when opening the dxf file.

svenhb, 2020-07-29:
There was a bug in the implementation of #132 
Please check new release

lianzaozi, 2020-07-31:
The cutting line of the software cutting function of version V1.5.0.3 is displayed normally, and the problem that the software cannot be used due to the wrong cutting distance set by the user has been solved.

svenhb, 2020-08-01:
Ok, so implementation seem to work. Issue can be closed

closed 2020-08-01 by svenhb