svenhb/GRBL-Plotter #110: "Preview" button.

Phoenix-I, 2020-03-05:
Hello.
The offer to make the "Preview" () button. Like on industrial controllers. The function passes a rectangle along the extreme dimensions of the cut.
We calculate the leftmost X, the rightmost X, the extreme Y - we build a rectangle.
It would be very convenient and useful, knowing whether there is enough space on the workpiece during positioning.

Thanks in advance.

svenhb, 2020-03-05:
Just a surrounding box, around the tool-path?
I will implement soon.

Phoenix-I, 2020-03-07:
This is usually the surrounding box.  
This is much easier to implement.

2 Exampls. 
Green is the preview path.
Blue is a work figure.

[image]

Phoenix-I, 2020-03-07:
Sorry :) I don’t often get to the hub.

There is still a thought, but I think it will be difficult.

Recognition of nested shapes and cutting out not in the order of creation, for example, like here 1, 2, 3, 4, but by nesting. 4, 3, 2, 1.

If by creation - a square falls out on the laser or milling cutter, and there are no other figures.
[image]

I'll go update the program and try the "Preview".

Thanks a lot. :)))

svenhb, 2020-03-08:
Added dimension-box and new sort-option for groups.
https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.3.4.0
Give your objects different colors, to get different groups.

lianzaozi, 2020-03-09:
Version 1.3.4.0 encountered a problem when simulating the trajectory of a dxf file. The circles in the dxf file will always be simulated.

The file test20.zip is the dxf file I used for testing
[image]
[test20.zip](https://github.com/svenhb/GRBL-Plotter/files/4305277/test20.zip)

lianzaozi, 2020-03-09:
The source code of version 1.3.4.0 should be changed to the same as shown in the figure, otherwise, running the code will give an error.
[image]

svenhb, 2020-03-09:
Check bug fix for endless simulation https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1341_publish.zip
I don't think class renaming is needed, perhaps you need to rebuild complete code?
It's working fine for me...

lianzaozi, 2020-03-10:
The simulation function of version 1.3.4.1 works very well, thank you very much.

Phoenix-I, 2020-03-10:
It seems we did not understand each other :)), I poorly explained.

The surrounding box around the work area is good.

How can I get the laser head to pass along it (the path along the lines of the box, the laser is turned off)?

The principle is this:
1. I put plywood.
2. I load the drawing. 3. I will identify the 0 coordinate.

Now the question is, will it fit or not?

4. I start the preview, the tool goes through the extreme points - I understand.

If intermeddle - burn.
If it doesn’t interfere, I’m either menu item 0 or I’ll take another blank.

Who uses a laser diode - can include a minimum power as a backlight.
I switched from diode to CO, and I use a simple red laser diode as a pointer to the place of cut - it is always on.

svenhb, 2020-03-10:
I see, I will add a menu entry in "Workpiece". 
Rapid move arround the surrounding box would be ok, I think.

Phoenix-I, 2020-03-10:
This is usually a hotkey, right on the panel.
Often used several times before starting.
I would make it a free button from the top right Bt9 or Bt10.

svenhb, 2020-03-12:
I add new variables and a script.
Just apply the script "script_move_arround_dimension.nc" to a button to move arround the gray dimension box.

lianzaozi, 2020-03-14:
> 我添加了新变量和脚本。
> 只需将脚本“ script_move_arround_dimension.nc”应用于按钮即可移动灰色尺寸框。
> [image]
> 
> https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.3.4.2

When the source code of version 1.3.4.2 is compiled and run, an error will be reported. It should be something you ignored when uploading. You can download the code you uploaded from github and try it on your computer.

svenhb, 2020-03-14:
The source code doesn't contain the latest 
_App.config_ and 
_GRBL-Plotter.csproj_ files... 
Forgot to upload them before creating the releae.    
But I uploaded them here https://github.com/svenhb/GRBL-Plotter/tree/master/GRBL-Plotter

lianzaozi, 2020-03-14:
Thank you for your reply.
I downloaded the code from the link you provided, but I still get an error when compiling and running.

lianzaozi, 2020-03-14:
> 源代码不包含最新的
> _App.config_和
> _GRBL-Plotter.csproj_文件... 
> 在创建版本之前忘记上传它们。
> 但是我在这里上传了它们https://github.com/svenhb/GRBL-Plotter/tree/master/GRBL-Plotter

thank you for your reply.
I downloaded the code from the link you provided, but I still get an error when compiling and running.

lianzaozi, 2020-03-14:
> The source code doesn't contain the latest
> _App.config_ and
> _GRBL-Plotter.csproj_ files...
> Forgot to upload them before creating the releae.
> But I uploaded them here https://github.com/svenhb/GRBL-Plotter/tree/master/GRBL-Plotter

Thank you for your reply.
I downloaded the code from the link you provided, but I still get an error when compiling and running.

svenhb, 2020-03-14:
I uploaded some files again, check for Version 1.3.4.2

lianzaozi, 2020-03-14:
> I uploaded some files again, check for Version 1.3.4.2

The problem still exists, you can download the entire project you uploaded and test it on your own computer.

svenhb, 2020-03-14:
Ok, I uploaded now ControlSerialForm, ControlCameraForm and data/scripts/...move_arround

lianzaozi, 2020-03-14:
> Ok, I uploaded now ControlSerialForm, ControlCameraForm and data/scripts/...move_arround

Very good, the problem has been resolved.

svenhb, 2020-03-14:
Very good.
Thanks for feedback

svenhb, 2020-03-22:
* 2D-View: fix strange zomm-in, -out behavior  
Reset zooming on double-click middle mouse button    

https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.3.4.3

Hopefully I uploaded all needed files this time

closed 2020-03-22 by svenhb