svenhb/GRBL-Plotter #111: Suggestions for Improvement

lianzaozi, 2020-03-18:
I think that when you choose not to load any height map file, this pop-up window may not be needed, including the pop-up window that reminds you if you need to close the software. Do these pop-ups have other intentions? Maybe I haven't understood the function of these pop-ups?

[image]

svenhb, 2020-03-18:
Ups, it's a bug. The open dialog also triggers the main window to show the height-map (if already GCode is displayed). But if you cancel the open dialog, there is no valid data in the map variable to show - this throws an unhandled exception, which I handle with the "Quit GRBL-Plotter question"...
  
Also the Text on the blue button "Apply Height Map" "Remove Height Map" seems to be wrong some times....

svenhb, 2020-03-22:
Bug is fixed    
* 2D-View: fix strange zomm-in, -out behavior  
Reset zooming on double-click middle mouse button    

https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.3.4.3

closed 2020-03-22 by svenhb