svenhb/GRBL-Plotter #112: Laser cutting path planning

lianzaozi, 2020-03-25:
First of all, thank you for your commitment to the plotter software.

I have an idea. I want to use this software for laser cutting. Generally, laser cutting has to plan the path first, and then read the dxf file to generate a specific trajectory Gcode. I looked for it in the software and did not find a setting item about the path planning of the laser cutting.

svenhb, 2020-03-25:
You mean planning to reduce unneeded G0 moves?
Including reversing of lines (switching start- and end-point)?

lianzaozi, 2020-03-25:
thank you for your reply

What I mean is not what you understand.

Let me give an example. Take the following picture as an example. Figure 3 is the product I want to actually get after laser cutting. I used dxf in Figure 1 to import the software. After the software actually planned the path, the starting point of the cut was A and B, because the corners of the cut product would fit the size. The cutting start points A and B and the green segment line trajectories in Figure 2 are automatically planned by the software according to the cutting process settings. The red arrow in Figure 2 is the motion trajectory. (I remember FastCAM software has similar nesting function and path planning function)

[image]
Figure 1

[image]
Figure 2

[image]
Figure 3

svenhb, 2020-03-25:
I think this is too complicated for me.
Perhaps you can do the filling in Inkscape using the Egg-Bot add-ons?
https://wiki.evilmadscientist.com/Hatch_fill

closed 2020-04-14 by svenhb