svenhb/GRBL-Plotter #113: About the method of quickly adjusting the trajectory order

lianzaozi, 2020-03-25:
If I want to adjust the trajectory order, now click on the Gcode generated by the geometric primitives in the Gcode display area and select "Code Blocks". In the pop-up menu, "Move selected code block up" and "Move selected code block Down" cannot be used. Or you can use Gcode blocks and use shortcut keys to adjust the order.
As shown in Figure 1
[image]
 Figure 1

If I can drag the selected Gcode up or down to adjust the order, I think it will be great. As shown in Figure 3, the selected part can be dragged to the position of the red arrow with the mouse.
[image]
 Figure 3

The first line of the gCode block generated by each geometric element is added to display the name of this geometric element as shown in Figure 2. In this way, if there are many geometric elements, from the gcode display area can help me determine the corresponding gcode block. Which geometric primitive. Of course, directly clicking on the geometric primitive will also highlight the corresponding gcode, which is indeed possible but not convenient. Now directly clicking on the geometric primitive will expand and display the gcode block corresponding to the selected primitive. Sometimes I just select the primitive to see which corresponding Gcode block is. It is not like expanding the gcode block to display it. I hope I can also add When selecting whether the gcode block corresponding to the geometric primitive is collapsed or expanded.
[image]
 Figure 2

svenhb, 2020-03-26:
Move selected code was made for "Groups", I will extend for any block.
I will also try:
- move up/down by one or on top
- Click in 2D-View will not expand a block (perhaps something like Shift + Click will expand)
- perhaps selectable XML-Attributes for the "Figure" Element
- perhaps sorting GCode-Blocks by attributes

svenhb, 2020-04-10:
Hi, check new version https://github.com/svenhb/GRBL-Plotter/releases/latest
With improoved Code-Blocks handling.

lianzaozi, 2020-04-11:
> Hi, check new version https://github.com/svenhb/GRBL-Plotter/releases/latest
> With improoved Code-Blocks handling.

how to use

svenhb, 2020-04-11:
https://github.com/svenhb/GRBL-Plotter/wiki#code-blocks

svenhb, 2020-04-14:
Hope the sort function for the code-blocks works for you?

closed 2020-04-14 by svenhb