svenhb/GRBL-Plotter #114: I found two problems in the GCode display area

lianzaozi, 2020-03-26:
I found two problems in the GCode display area

1. When I expand a GCode block and want to edit one of the lines, I find that as long as I position the cursor in the expanded GCode block, whether I add a new line / delete a line / change the content, then expanding the GCode block will automatically restart Folding, I think that every time automatic folding will cause trouble to the editor, because this will cause me to enter the content normally. Picture 1 is a screen recording gif file that I encountered when editing GCode.

2. Every time I type anything, every time I type a letter or number, the insertion cursor will automatically jump to other places. I don't think the cursor can change the position randomly, because it will cause me to input something incorrectly. The operation demonstration is in the second half of the gif file.

[image]

svenhb, 2020-03-26:
I will try to improove.
Thanks for your GIF-animations - how do generate them?

lianzaozi, 2020-03-26:
I use "ScreenToGif" software to record the screen first, and then save it as a gif file.

svenhb, 2020-04-10:
Should be solved by using "Edit Mode"

lianzaozi, 2020-04-11:
> Should be solved by using "Edit Mode"
> https://github.com/svenhb/GRBL-Plotter/releases/latest

Where is "Edit Mode"?

svenhb, 2020-04-11:
[image]

lianzaozi, 2020-04-11:
In "Edit Mode", when the mouse cursor is positioned at the X or Y coordinate of a line, the cursor automatically moves to the beginning of the gcode line. I cannot position the cursor where I want

svenhb, 2020-04-11:
Yes, thats right, because..... If I want to insert code, it should happen at the beginning of the line.
You can use left / right arrows of key board to adjust position...

lianzaozi, 2020-04-13:
> Yes, thats right, because..... If I want to insert code, it should happen at the beginning of the line.
> You can use left / right arrows of key board to adjust position...

Using the left and right keyboards to position the cursor is quicker than mouse clicks, but the method you said can also be used.

I have an idea that the track that has been moved during offline simulation is displayed in a different color. The color of the track that has no movement is different from the color of the track that has been moved, so that it can be more clearly determined which tracks have been moved.

svenhb, 2020-04-14:
Perhaps it's solved now
https://github.com/svenhb/GRBL-Plotter/releases

closed 2020-04-14 by svenhb