svenhb/GRBL-Plotter #116: A axis <360

wwwlxs, 2020-04-11:
Hi svenHB，Can you add the following function to the new version of grblplotter: If the angle change is greater than 360 degrees, the z axis will move up, the axis will rotate -360 degrees, and then move down to return to the original angle, for example, when the A axis is 361, will  Return 1 degree. This function can prevent the dispenser's trachea from rotating too many times and causing a break ,i have leave a message to your instagram.Best wishes !

[image]

[image]

svenhb, 2020-04-11:
I see.
How to handle an exceed of 360° during a move? Interrupt or finish the move, then 'unwrap'?

wwwlxs, 2020-04-11:
Interrupt. turn back to the same angle (for example, if -361degrees, then +360, turn to -1degrees, if 361degrees , then -360, turn to 1degrees, that is, the rotation is controlled within + -360 degrees,) avoid too much rotation too many turns, after completing the angle rotation, continue to move forward

lianzaozi, 2020-04-14:
You said that this function is very good, I think when the rotation angle is greater than 360, it is a good idea whether the Z axis rises and turns. This function should be made into an option for users to choose. Because some users have pneumatic slip rings installed on their machines, the pneumatic slip rings can prevent the trachea from being entangled when the shaft rotates more than 360 °. Such users do not need to turn on the function of Z-axis rising and turning when it is greater than 360 °.

svenhb, 2020-04-14:
I built in such option, hope it works as expected
https://github.com/svenhb/GRBL-Plotter/releases

closed 2020-04-15 by svenhb