svenhb/GRBL-Plotter #117: Version 1.3.4.5 Some of the track colors that have been moved will not change.

lianzaozi, 2020-04-15:
Thank you for your hard work. The function that the mouse can locate and modify the position in GCode editing mode added by the plotter service software in version 1.3.4.5 is great.

In version 1.3.4.5, the color of some motion tracks will not change. The following is the dxf document I used for testing.
[image]
[2020_test.zip](https://github.com/svenhb/GRBL-Plotter/files/4478617/2020_test.zip)

svenhb, 2020-04-15:
Yes, I know, and as faster the simulation as more red paths you will see.
Hard to fix and not really important I think.

lianzaozi, 2020-04-15:
Even if I reduce the movement speed to the slowest, there will still be movement of the track without changing the color. This phenomenon can be found from the demonstration of 2222.gif. In Figure 0000, I marked the place where the color has not changed.
The continuous trajectory movement should always follow the continuous trajectory until the movement reaches the end of the continuous trajectory before G00 moves to the next geometric primitive. But the animation demonstration in 2222.gif is not like this, and this movement is not the shortest path.
I think the correct approach should be to move in the direction indicated by the arrow in Figure 2222

[image]
Figure 0000

[image]
Figure 2222

[image]
1111gif

lianzaozi, 2020-04-15:
This is my test dxf.
[Drawing2.zip](https://github.com/svenhb/GRBL-Plotter/files/4482263/Drawing2.zip)

svenhb, 2020-04-15:
I import the elements in the order and direction as they are stored inside the original DXF file.
In LibreCad you can select all elements and "Create Polyline from existing segments". Then the elements are arranged as you want...
[image]

svenhb, 2020-04-15:
Try new release to fix gaps in already simulated path.
https://github.com/svenhb/GRBL-Plotter/releases

lianzaozi, 2020-04-16:
In the new version 1.3.4.6, there will still be no change in the color of the moving geometric trajectory.
Adjusting the step size according to your needs is very good. I have tested this function.

svenhb, 2020-04-16:
I see, I didn't notice this before...
G2/G3 commands will not be marked as simulated...

lianzaozi, 2020-04-16:
The semicircle in Drawing2 generates G01 (not G02 and G03), but there will be no change in the color of the trajectories after the movement of G01. Of course, simulating G02 and G03 will not change the color.

svenhb, 2020-07-28:
Does it still appear? I don't think so...

lianzaozi, 2020-07-31:
The software version V1.5.0.3 can simulate G02/G03 very well.

lianzaozi, 2020-07-31:
One thing I don’t quite understand. I would like to ask you. Why does the simulation speed increase, and the simulated lines of some geometric primitives do not change colors?

svenhb, 2020-07-31:
Thank you for testing.
>Why does the simulation speed increase, and the simulated lines of some geometric primitives do not change colors?  

I have to calculate the intermediate steps on G0, G1, G2 and G3 commands, the to show the time progress. Sometimes the last intermadiate step to finish a line or circle is skipped - should not happen and I can't find the reason.   For me it is not really important...

closed 2020-07-31 by svenhb