svenhb/GRBL-Plotter #118: Semicircle in DXF file cannot be parsed

lianzaozi, 2020-04-15:
Semicircle in DXF file cannot be parsed
Figure 1 is displayed by the plotter software
Figure 2 is a CAD file display
"Drawing1.zip" is the dxf file I used for testing
There are the same problems when testing with software versions 1.3.4.4 and 1.3.4.5.

Figure 1
[image]

Figure 2 
[image]

dxf file
[Drawing1.zip](https://github.com/svenhb/GRBL-Plotter/files/4479188/Drawing1.zip)

lianzaozi, 2020-04-15:
I haven't tested the software version 1.3.4.4 and 1.3.4.5.

lianzaozi, 2020-04-15:
I used another dxf file with a semicircle to test, and the semicircle in the other dxf file can be displayed. The arc in the file "Drawing1" cannot be displayed. But the problem must not be with the Drawing1.dxf file.

Drawing2 files can be displayed normally.
[Drawing2.zip](https://github.com/svenhb/GRBL-Plotter/files/4479589/Drawing2.zip)

svenhb, 2020-04-15:
Drawing1 is very small - default setting of 1.00 for step witdh is too low, better use 0.1:
[image]

lianzaozi, 2020-04-15:
I'm very sorry. I didn't pay attention to this option before. I tested it according to your method, and the semicircular arc was displayed.

svenhb, 2020-04-15:
No problem, I will add a detection to correct the "Arc step width" if needed (if radius is too small).
Thanks for your hard testing :-)

closed 2020-04-15 by svenhb