svenhb/GRBL-Plotter #119: Sorting and shortest path optimization

lianzaozi, 2020-04-19:
After comparative testing, I found that the plotter software does not have the shortest path optimization function. The current sorting is based on the order of reading primitives in the dxf document. The order of storing geometric primitives in the dxf document is only the order of primitive drawing. If the path planning is done in the order of geometric primitive drawing, this will result in a very long empty trip. The path planning that you said with libreCAD is a way. However, when the number of geometric primitives in dxf is relatively large, it is very difficult to use libreCAD for path planning, so I think that the addition of the shortest path planning function in the plotter software can bring great convenience, and can minimize the empty walk The time of the trip.

svenhb, 2020-04-21:
Yes, you are right.
Sorting by shortest distance I implemented yet (not released).
I'm not sure if it is ok to revers drawing direction of some primitives to optimize path lengths a bit more. For pen-plotters it's probably ok, for routers may be not...

lianzaozi, 2020-04-21:
The shortest path effect in the open source software dxf2gcode is excellent, you can learn from the algorithm code. I know that some people use Dijkstra's algorithm as the shortest path algorithm.

lianzaozi, 2020-04-21:
Generally speaking, the direction can be reversed during the shortest sorting process, which generally has no effect. However, some processing occasions must be processed in a certain order. In fact, you can refer to the method of dxf2gcode. After sorting, you can also allow users to adjust the direction of the geometric primitives.

lianzaozi, 2020-04-23:
Hello.
I think that when the software reads the dxf file, the software sorts the shortest path of the read geometric primitives according to the algorithm. After the sorting is completed, if the user feels that the sorting result of some geometric primitives is not the result he wants, he can allow the user to adjust the direction of the primitives and the order of certain primitives. The shortest path sorting can be used as the default sorting method of the software. At present, the software has the sorting order according to the drawing order of geometric primitives as an option for the user to choose, which will be better.

svenhb, 2020-04-23:
The sorting method on dxf2gcode looks to complicated to me (it's using TSP).
Reversing the order of geometric primitives will not work for tangential axis and perhaps not work for other special use cases - if done G-Code level.
Perhaps I need to implement a intermediate level to be able to reverse the order of primitives, before translating the "Pen up/down" possibilities.

lianzaozi, 2020-05-02:
Genetic Algorithm TSP

This is an experiment of applying Genetic Algorithm to Travelling Salesman Problem, as well as visualizing the algorithm.

Applying Genetic Algorithm to Travelling Salesman Problem  http://parano.github.io/GeneticAlgorithm-TSP/

svenhb, 2020-07-23:
Check [Sort objects](https://github.com/svenhb/GRBL-Plotter/wiki/Import-options#graphics-import-general) option in new release

lianzaozi, 2020-07-24:
The newly added function of sorting according to the shortest path in the software, I think the effect is very good after testing.

lianzaozi, 2020-07-24:
I think the shortest path sorting algorithm needs further optimization. In the screenshot below, I think it will take less time to start and finish processing from the position pointed by the blue arrow. Perhaps the current software algorithm may have considered other reasons that I have not considered, leading to the current shortest path sorting method.

[image]

This is the dxf file I used for testing

[ellipse.zip](https://github.com/svenhb/GRBL-Plotter/files/4972593/ellipse.zip)

svenhb, 2020-07-26:
Should be fixed in 1.5.0.1 Check new option 
[image]

lianzaozi, 2020-07-27:
Now check the newly added function option "Allow new start point in closed path" The shortest path sorting effect of the ellipse is very good

I used the circle test and found that the shortest path sorting effect is the best if the circle is subdivided into small straight lines.

When the GCode generated by setting the circle is G02 and G03, the sort result
[image]
[image]

When the GCode generated by setting the circle is G01, the sort result
[image]
[image]

svenhb, 2020-07-27:
I made an extra function to handle full circle made of G2/G3 to move the start-point arround the circumfence. In my example it worked, in yours not - I will check.

svenhb, 2020-07-28:
Check new release for fixed start point in G2 G3

lianzaozi, 2020-07-29:
Version V1.5.0.2 of the software will cause some graphics elements in dxf to fail to display. Is the reason related to the newly added "correct start point within G2 / G3 circle" function?

The screenshot image1 below is a screenshot of the software version V1.5.0.2
[image]

image2 is a screenshot of the software version V1.5.0.1
[image]

[image]
setup
[image]

svenhb, 2020-07-29:
There was a bug in the implementation of #132 
Please check new release

lianzaozi, 2020-07-30:
Thank you very much, now the circle display is normal, and the software has a good effect on the shortest path planning of the circle.

svenhb, 2020-07-30:
You need to check the yellow info-line after import....  
Some options will check "Avoid G2/G3" internally - like clipping. 
The applied options can be seen in the yellow line:  
[image]

lianzaozi, 2020-07-30:
Sorry, it was my mistake. I activated the clipping function, which caused the circle and arc to generate G01. This is not a software error.

The circle sorting of version V1.5.0.3 works well

svenhb, 2020-07-31:
Thanks for testing

closed 2020-07-31 by svenhb