svenhb/GRBL-Plotter #120: SVG Import - First Figure/Path

LukasFritz, 2020-05-14:
The first figure after Converting SVG to GCode has no `Z0.0` Command. The Plotter dragknife starts plot on my save hight. The second figure has the Command `G01 Z0.000 F1000`. It starts at my Pltting hight and work very well. I have testes multiple SVG files.

Nochmals in Deutsch.
Als bei dem ersten Pfad meiner SVG wird im G-Code kein Z0.0 Kommando ausgeführt, folglich plottet er auf der Sicheren Höhe bei Z4.0 erst mit dem Beginn des 2. Pfads klappt alles wie gewohnt.

Anbei die SVG und das GCode File. Ich benutze Version 1.3.4.6

[Test4Eck_.zip](https://github.com/svenhb/GRBL-Plotter/files/4626767/Test4Eck_.zip)

svenhb, 2020-05-14:
Does it also happen after reloading the same graphics?

LukasFritz, 2020-06-06:
yes

svenhb, 2020-06-25:
The problem is the value of "Z-Height engrave" = 0.00 - it should be negative... 
[image]

I will change the possible values for this control...

svenhb, 2020-07-23:
Check new release if fixed

closed 2020-07-26 by svenhb