svenhb/GRBL-Plotter #121: Dxf file parsing error

lianzaozi, 2020-05-14:
Hi, I recently discovered a problem while using plotter software, and I do n’t know how to solve it. When I use the software to load the dxf file, the software automatically adds some straight lines, but these straight lines are not in the dxf file. The following Figure 1 is a screenshot of a dxf document opened with Estlcam software, Figure 2 is a screenshot of a dxf document opened with a plotter software, and Figure 3 is a screenshot of a dxf document opened with LibrCAD software. The part marked with blue in Figure 2 is added automatically by the plotter software.

[image]
Figure 1 

[image]
Figure 2

[image]
Figure 3

Dxf file for testing
[temple_noir_aplats_negatif.zip](https://github.com/svenhb/GRBL-Plotter/files/4631036/temple_noir_aplats_negatif.zip)

svenhb, 2020-05-15:
Did you use special options or did you sort the figures after importing?

lianzaozi, 2020-05-15:
thank you for your reply.
I made the gif animation of the process of reading dxf with plotter software and the process of reading the same dxf file with librecad software.

[image]

lianzaozi, 2020-05-18:
Sorry, because of my mistakes, I did not check whether the primitives in the dxf file coincide, in fact, there are many coincident geometric primitives in the dxf file used for testing, and these primitives are very short. Therefore, the extra lines when reading the dxf file with the plotter software should be caused by the overlapping geometric primitives in the dxf file. But I have a question, even if there are overlapping geometric primitives in dxf, the lines that were not in the dxf file should not appear. If there are repeated geometric primitives, it should be repeated geometric primitives to generate Gcode twice.

Based on the above, I have two good suggestions:
1. The plotter software can delete duplicate geometric primitives (completely repeated geometric primitives will delete one of them, and when a short primitive and a long primitive coincide, delete the short one)
2. When the distance between two straight line primitives is less than or equal to the set allowable value, merge the two primitives.

To illustrate my idea, I drew three pictures
Figure 1: If the red geometric primitive and white geometric primitive coincide, delete the short red geometric primitive.
[image]

Figure 3: If the two geometric primitives partially overlap, the two geometric primitives are merged into one primitive. The total length of the merged geometric primitives is the length of the two repeated geometric primitives minus two The length of the part where the geometric primitives coincide.
[image]

Figure 2: When the distance between two primitives is less than the set allowable distance, it intersects with the extension of the two primitives. Among them, the extended part of the geometric element with an angle can be selected to be a straight line connection or a circular arc connection (when the circular arc connection is selected, this arc should be tangent to two straight lines).
[image]

I do n’t know if I express your understanding of my idea like this. I know that it is difficult to implement, but it does prevent some strange problems when reading dxf files.

lianzaozi, 2020-05-18:
When looking at pictures, please double-click the picture you want to view, so that you will see more clearly.

svenhb, 2020-07-23:
Check [Distance to be assumed as equal](https://github.com/svenhb/GRBL-Plotter/wiki/Import-options#graphics-import-general) in new release

lianzaozi, 2020-07-27:
Thank you very much. After adding this function, the plotter software will not generate some lines that did not exist when reading the "Temple_noir_aplats_negatif.dxf" file. But my use and understanding of this function is still not very clear.
I am not very clear about the use of "Distance to be assumed equal", can you give a detailed introduction to the use of this setting item and the function of this setting item?

svenhb, 2020-07-27:
Usually you would compare coordinates like 
_if ((x1 == x2) && (y1 == y2))_, but with float and double format there is a risk of minimal deviation even if numbers are 'same', perhaps caused by rounding.    
 Better work with a 'range': 
_if ( ( Math.Abs(x1 - x2) < deviation) && ( Math.Abs(y1 - y2) < deviation) )_   
With 'deviation' ("Distance to be assumed equal") you can set the allowed error...

closed 2020-07-30 by svenhb