svenhb/GRBL-Plotter #122: Shortcut key for empty track

lianzaozi, 2020-05-14:
I have a small suggestion on whether I can modify the shortcut key for displaying the empty walk track to display the empty walk track by pressing the space bar, and hide the empty walk track by pressing the space bar again. This will be more convenient. The current operation method of the software is that only when you keep pressing the space bar will the track be hidden.

svenhb, 2020-05-15:
Good idea. I will implement.

lianzaozi, 2020-05-15:
I suddenly thought that the space bar will also be used when editing gcode in the gCode display area, so when editing Gcode, it may affect the display of the empty travel track. If it is really influential, you can change the shortcut key that controls the display of the empty travel track to another key

svenhb, 2020-05-24:
Toggling the pen-up path only appears if the space-bar is pressed when the 2D view has the focus.

lianzaozi, 2020-05-25:
OK, I get it. thank you for your reply. Hope the plotter software is getting better and better.

svenhb, 2020-07-23:
Check new release

lianzaozi, 2020-07-24:
After the modification, it is very convenient to show and hide the empty track, thank you.

closed 2020-07-26 by svenhb