svenhb/GRBL-Plotter #123: Will the plotter software add the function of generating two-dimensional codes to one-dimensional codes in the future?

lianzaozi, 2020-05-24:
Will the plotter software add the function of generating two-dimensional codes to one-dimensional codes in the future?
If the software provides such a function, you can directly use the plotter software to draw the QR code or use the laser to engrave the QR code. Some open source QR code 1D code libraries can be used directly on github. QRCoder (https://github.com/codebude/QRCoder)
QR.NET (https://github.com/aaronogan/QR.NET)
ThoughtWorks.QRCode (http://www.codeproject.com/Articles/20574/Open-Source-QRCode-Library https://github.com/aaronogan/QR.NET)
QRCodeTools (https://github.com/yunfeifei/QRCodeTools)
ZXing
ZBar

svenhb, 2020-05-24:
It may work to copy the QR image to the clipboard and paste it in GRBL-Plotter.
Or do you need the generation inside GRBL-Plotter?

lianzaozi, 2020-05-25:
External generation is a way of importing into the plotter software.
I think the plotter software directly integrates the function of generating gcode from two-dimensional codes and one-dimensional codes, which is very convenient to use. Just as the plotter software now supports generating gcode from text / pictures / simple shapes, it is very convenient.

lianzaozi, 2020-05-26:
ZXing.Net is an open source library that supports two-dimensional codes and one-dimensional codes very well, and can be called by c # language. It can be found on github.

svenhb, 2020-07-23:
Check menu 'G-Code creation - Create Barcode'
[image]

closed 2020-07-26 by svenhb