svenhb/GRBL-Plotter #124: Outline word filling

lianzaozi, 2020-05-25:
Recently, when I used the plotter software, I found that there are two fonts built in the plotter software: 1. Single-line font 2. Outline font

There are only 4 fonts that can display Chinese: unicode.lff, kst32b.lff, azomix_i.lff, azomix.lff. These 4 fonts are single-line fonts, which I like very much, but they can only support some Chinese fonts. This leads to the failure to create correct gcode when the Chinese I input is a font that is not supported. I searched through Google for a font that supports Chinese comprehensively. This font is wqy-unicode.lff. The Chinese I entered using this font can be displayed correctly. But this font is outline font. Of course, outline font is very helpful for laser cutting, so laser cutting uses outline font. But if I want to use laser to engrave the characters, I often use it to fill the font. So I hope that the interface of generating gcode from text can add the option of outline word filling. In this way, I can use laser engraving or use a pen to write the effect of filling the font.

Figure 1 is the effect of filling the outline word
[image]

Figure 2 is the processing result of the contour word by the plotter software
[image]

svenhb, 2020-07-23:
Check new option [Hatch fill](https://github.com/svenhb/GRBL-Plotter/wiki/Import-options#fill-closed-paths) in new release.
[image]

lianzaozi, 2020-07-24:
The outline word filling function is very easy to use and great.

closed 2020-07-26 by svenhb