svenhb/GRBL-Plotter #125: Overlay fonts on 5k displays

wwwlxs, 2020-06-23:
Overlay fonts on 5k displays
[image]

wwwlxs, 2020-06-23:
Some Chinese translations are not very accurate, I can help improve the translation

svenhb, 2020-06-23:
Wow this is huge :-) But I don't see the bug... 
Are the markers for tool position too big? 
Or should the virtual joystick be larger?
Or the editor?:
[image]
You can use Ctrl+Mousewheel to enlarge editor font - perhaps this helps?

>Some Chinese translations are not very accurate, I can help improve the translation

Yes this would be nice - perhaps deliver screenshot (so I know where) and Text to copy&paste for easy replacement.

wwwlxs, 2020-06-24:
1 It is the editor ，Ctrl+Mousewheel can only scroll up and down，cannot to enlarge editor font on windows10 with iMac.  2 The black comment in the pdf file is ok, the gray is not sure.I don’t know how to use 'Tool change' , so I didn’t modify the translation “how does it work”in Tool change
[grblplotter simplified chinese .zip](https://github.com/svenhb/GRBL-Plotter/files/4827152/grblplotter.simplified.chinese.zip)

wwwlxs, 2020-06-24:
Another bug?
[image]

wwwlxs, 2020-06-27:
[image]

svenhb, 2020-07-23:
Check new setting line distance in new release:
[image]

svenhb, 2020-07-31:
I assume it is working as expected

closed 2020-07-31 by svenhb