svenhb/GRBL-Plotter #126: Add Pos.A to Tooltable

wwwlxs, 2020-06-27:
Hi
Can add Pos.A to Tooltable  in the new version ?My plotter is 4-axis and is placed vertically.I want to make it to a “WaterColorBot” like your YouTube vedio，and How to setup “ToolChange” and “Tooltable” and “G-code modification”？
[image]
[image]

svenhb, 2020-06-27:
>How to setup “ToolChange” and “Tooltable” and “G-code modification”？  

Not so easy to describe:
1. tools will be seleceted by graphic colors: check video https://www.youtube.com/watch?v=Ebe2kFlE058  at 0:10  
in the video green is assigned for a pen, red / dash line for weak laser, black for strong laser for cutting.
2. Enable Grouping (Graphic Import - Path Import)
3. Add tool exchange command (Graphic Import - Add tool change command)
4. To interrupt drawing to refresh water-color you may want:
[image]
5. Setup your script in 4) for your needs.

>Can add Pos.A to Tooltable in the new version?  

I don't understand how do you select a color. I will insert a a position for tool change position...

In new version (I still need some weeks) tool can also be selected by 
tool-width <-> pen-width
tool-name <-> Graphic-Layer

wwwlxs, 2020-06-28:
> I don't understand how do you select a color. I will insert a a position for tool change position...
this video shows:
[我的项目 81.MP4.zip](https://github.com/svenhb/GRBL-Plotter/files/4842105/81.MP4.zip)

wwwlxs, 2020-06-28:
another bug:The simulation preview and actual streaming preview are show different
[image]

[image]
[压缩文件.zip](https://github.com/svenhb/GRBL-Plotter/files/4842118/default.zip)

wwwlxs, 2020-06-28:
another bug? 
this code cannot run in the ZX plane and becomes a circle in the XY plane
`G91 G01 X-8.000 Z0.000 F4000    ( stir brush inside cup )
G02 X0.000 Z0.000 I8.000 K0.000 
G02 X0.000 Z0.000 I8.000 K0.000 
G02 X0.000 Z0.000 I8.000 K0.000 
`

svenhb, 2020-06-28:
GCode manual says you need to switch the plane via G17, G18 or G19 - not sure if grbl supports this: http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g2-g3

wwwlxs, 2020-06-28:
ok，thank you very much，it supported in V1.1 https://github.com/gnea/grbl/wiki

svenhb, 2020-07-23:
Check new release, unfortunatly not really tested, I hope it works.
[image]

wwwlxs, 2020-07-23:
ok, thanks again ,I will check it (maybe tomorrow )
 ,my plotter has some problems now

closed 2020-08-15 by svenhb