svenhb/GRBL-Plotter #127: SVG Line stroke-width map to Z value

wwwlxs, 2020-07-03:
Hi sven, Feature request: Line stroke-width map to Z value(Z-Min , Z-Max)(or constrain Z-Min & Z-Max )
[image]
[image]
[line.svg.zip](https://github.com/svenhb/GRBL-Plotter/files/4869404/line.svg.zip)

svenhb, 2020-07-03:
Yes, I was also thinking about that, but could not generate data for this. I planned to support a graphic-tablet and take drawing data (including pen-pressuse) inside GRBL-Plotter, but I couldnt find example code for windows-form application to include graphic-tablet (examples are always in WPF)...
Anyway I will check how I could support that.

wwwlxs, 2020-07-03:
> Yes, I was also thinking about that, but could not generate data for this. I planned to support a graphic-tablet and take drawing data (including pen-pressuse) inside GRBL-Plotter, but I couldnt [...]

this line was draw from iPad and apple pencil ,  pen-pressuse and svg files was modified  for gcode,

wwwlxs, 2020-07-03:
or Feature request:Can GCode (with different Z values) be imported into GRBL-Poltter just like SVG is imported into GRBL-Poltter to automatically add A value?

svenhb, 2020-07-03:
>or Feature request:Can GCode (with different" Z values" by pen-pressure) be imported into GRBL-Poltter just like SVG is imported into GRBL-Poltter for automatically add" A value"?

I don't understand. You can replace 'Z' by 'A' using the editors-replace function (also available via menu). 
What do you mean with "for automatically add" A value" - from where should the 'A' value come?

wwwlxs, 2020-07-03:
> I don't understand. You can replace 'Z' by 'A' using the editors-replace function (also available via menu).
> What do you mean with "for automatically add" A value" - from where should the 'A' value [...]

for example,when I draw a curve in adobe illustrator and make GCode  ( with pen-pressure" Z values"  , But at this time there is still a lack of tangential value) with GCode plugin   ,then I import the GCode to GRBLPlotter for adding tangential axis (A value)and tool change ,then send it to plotter.
if I paste the curve from illustrator to GRBLPlotter,tangential axis value and tool change code can be add to GCode from 'G-Code modification 'setting  ,but lack of  "pen-pressure- Z values" .
 if I paste  GCode that exported  from adobe illustrator , all of" Pen-pressure-Z values" are works, but lack of  tangential axis value and tool change code.

svenhb, 2020-07-23:
Check new option [Translate pen with to Z value](https://github.com/svenhb/GRBL-Plotter/wiki/Import-options#path-interpretation) in new release.

wwwlxs, 2020-07-23:
> Check new option [Translate pen with to Z value](https://github.com/svenhb/GRBL-Plotter/wiki/Import-options#path-interpretation) in new release.

thanks, I will test it,There may be a bug in version 1.46. I wonder if it has been fixed?  That is, when" tangential axis" and "drag tool compensation" are used together, the angle of rotation and the new track is incorrect. If you create a square track, you can find it

svenhb, 2020-07-24:
>That is, when" tangential axis" and "drag tool compensation" are used together  
I don't think this is a usefull combination...
But this could be fixed because of "new backbone" during import and path modification.

wwwlxs, 2020-07-24:
Useful for correcting the position deviation of the tool（ brush ）and the axis

svenhb, 2020-07-24:
I checked the code - it will not work, because: 
1st I calculate the tangential angle of the original graphic
2nd I calculate new drag-knife path

For the next version I will switch the order - than it could work as expected.

wwwlxs, 2020-07-25:
HI svennb,In the same svg file, some lines have variable z values, but the z values ​​of some lines do not change with the line width.SVG file:
[A5.zip](https://github.com/svenhb/GRBL-Plotter/files/4974953/A5.zip)
The red marked lines have the same z value for each segment, and the black ones are variable
[image]

wwwlxs, 2020-07-26:
[image]

wwwlxs, 2020-07-26:
[image]
Will this （ green line）change with the z value of each line?  Or can it only be based on the setting of z-Height engrave?

wwwlxs, 2020-07-28:
[image]

svenhb, 2020-07-28:
Version 1) is my current implementation 
You mean like version 2) ?
[image]

svenhb, 2020-07-28:
From your post above:  
>Will this （ green line）change with the z value of each line? Or can it only be based on the setting of z-Height engrave?  
  
Nice graphics to show the relationship z-depth and radius - It will be calculated from the initial 'Z-Heigth engrave'

svenhb, 2020-07-28:
Check new release for missing z-depth - because of path reversing, z-data got lost

wwwlxs, 2020-07-29:
> From your post above:
> 
> > Will this （ green line）change with the z value of each line? Or can it only be based on the setting of z-Height engrave?
> 
> Nice graphics to show the relationship z-depth and radius - It will be calculated from the initial 'Z-Heigth engrave'

> Version 1) is my current implementation
> You mean like version 2) ?
> [image]

Yes, I mean that version 2 is reserved for the brush as the default function, which will suit the actual soft brush. If the length of each green line is very long, you can keep version1 as an option and use it for hard pen lines

wwwlxs, 2020-07-29:
> Check new release for missing z-depth - because of path reversing, z-data got lost

 z-data got lost:)

wwwlxs, 2020-07-29:
[H.zip](https://github.com/svenhb/GRBL-Plotter/files/4991852/H.zip)
I tested the svg file on V15.02 and it doesn't seem to work.  
"A generic error occurred in GDI+ "pops up, and the generated gcode is still the same as version 15.01

wwwlxs, 2020-07-29:
> From your post above:
> 
> > Will this （ green line）change with the z value of each line? Or can it only be based on the setting of z-Height engrave?
> 
> Nice graphics to show the relationship z-depth and radius - It will be calculated from the initial 'Z-Heigth engrave'

If the length of the green line can be changed with different Z values, it will be more accurate

svenhb, 2020-07-29:
[image]

New option to make a ramp instead of steps...
Check new release

wwwlxs, 2020-07-29:
> [image]
> 
> New option to make a ramp instead of steps...
> Check new release
Thank you very much, you are great! I test by this file: https://github.com/svenhb/GRBL-Plotter/files/4991852/H.zip,The new version is work!

wwwlxs, 2020-07-29:
> [image]
> 
> New option to make a ramp instead of steps...
> Check new release
But there are still some problems, the line direction is reversed, which is wrong for writing calligraphy

lianzaozi, 2020-07-31:
Why is the GCode generated by opening H.SVG different from the uploaded H.nc

image2 is a screenshot of my opening H.SVG
[image]

image3 is a screenshot of my opening H.nc
[image]

wwwlxs, 2020-07-31:
> Why is the GCode generated by opening H.SVG different from the uploaded H.nc
> 
> image2 is a screenshot of my opening H.SVG
> [image]
> 
> image3 is a screenshot of my opening H.nc
> [image]
Try to increase the range of “z-height engrave” 
[image]

svenhb, 2020-07-31:
I don't know when you saved the .nc file....
Gcode (like your H.nc) can not be modified with all the new options - it is only for import of SVG or DXF files....
And there is no chance for implementation.
The difference is because of the new option "Apply ramp instead of a step" to get the Z movement as shown in example 2).   

[image]

wwwlxs, 2020-08-01:
> And there is no chance for implementation.

hi svenhb， I uploaded the newest calligraphy test video to my instagram, you are welcome to take a look.if “drag tool compensation”can be calculated
 from pen width，it will be better. But it's pretty good now.thank you very much.
[image]

lianzaozi, 2020-08-01:
Thank you very much for your reply
I used the H.SVG file you uploaded, but I can't get the Z-axis coordinates of the gradient. I don't know why. From the gif animation, I can see that the Z-axis coordinates of the GCode I get are all -1. The whole process of operating the software can be seen in the gif animation

[image]

svenhb, 2020-08-01:
The maximum depth (or minimum Z value) is limited by the engrave depth - for safety reasons...
[image]

lianzaozi, 2020-08-01:
You are right, thank you, now you can get the Gcode where the value of the Z axis changes with the line width. After your guidance, the software is now working properly.

svenhb, 2020-08-04:
Drag tool % now checks actual depth

wwwlxs, 2020-08-04:
> w with Version

thanks！

wwwlxs, 2020-08-05:
> Before: "Drag tool Offset in % of Z-depth" used the "Z-heigth engrave" value.
> Now with Version 1.5.0.4 : it uses the Z-depth, calculated by penWith (if option is enbaled
> [image]

[image]
thanks you very much，It  work now！ Maybe there is a problem with my SVG file, or there is a problem with my settings, the lines are a bit jittery，I will try further and submit to you if there is any problem

svenhb, 2020-08-15:
Seems to work

closed 2020-08-15 by svenhb