svenhb/GRBL-Plotter #128: A axis and B axis can only use 1 axis, but display 2 axis

wwwlxs, 2020-07-04:
A axis and B axis can only use 1 axis, but display 2 axis
[image]
[image]
[image]

svenhb, 2020-07-04:
Your controller probably controls 5 axis - correct? Then A must be named "A":
And I think GRBL 4th axis can be switched off.  
[image]

wwwlxs, 2020-07-04:
yes,grbl-Mega-5X,5 axis,arduino Mega2560+ramps1.4

wwwlxs, 2020-07-04:
[image]

wwwlxs, 2020-07-04:
https://github.com/fra589/grbl-Mega-5X. and end stop doesn't  works

svenhb, 2020-07-23:
Does it work now? I think it is a settings issue.

wwwlxs, 2020-07-23:
It's not working for the time being, and it has not been retested. The homing function of grbl-mega-5X does not work

wwwlxs, 2020-07-26:
> Does it work now? I think it is a settings issue.

[image]
[image]

svenhb, 2020-07-26:
Should be fixed in 1.5.0.1 (except gamepad)

svenhb, 2020-08-15:
I will not change the GamePad dialog - the user has to be careful himself

closed 2020-08-15 by svenhb