svenhb/GRBL-Plotter #129: Questions about simulation and offset

lianzaozi, 2020-07-24:
Thank you for your hard work. This version brings exciting features. I like it very much. Below are the problems I encountered in my test (the software version used for the test is V1.5.0.0)

1. V1.5.0.0 version of the software can now simulate circles and arcs (G02 and G03), but the software will pop up an error when simulating circles and arcs (G02 and G03). As shown in Figure 1
[image]

2. After setting the offset, the position of the red graphic will change, but the position of the gray graphic will not change, and it will remain at the original position. as shown in picture 2
3. The gif file can show the whole process of my operation and the screen record of the problem.

[image]

git
[image]

lianzaozi, 2020-07-24:
During simulation, there is a pause at one point of the arc, and there is no pause at the other point. The phenomenon can be observed in the gif animation. Below is the dxf file I used for testing. The position indicated by the mouse in the 1234.gif file is the position where the simulation will pause.

[image]

There is a test dxf file that is paused during simulation
[2013.zip](https://github.com/svenhb/GRBL-Plotter/files/4970783/2013.zip)

Test dxf file for simulation and offset problems
[circle.zip](https://github.com/svenhb/GRBL-Plotter/files/4970788/circle.zip)

svenhb, 2020-07-24:
Strange: 2013.dxf is simulated correctly -  without error (which may causes the pause after each arc on your side).
Did you use special options? Perhaps you can check the logfile for more information?
In the "About" window you will find the startup path, where "logfile.txt" is located.

I used this options (using z axis) and get just one figure:
[image]

lianzaozi, 2020-07-24:
I probably found the reason for the pop-ups and pauses in the simulated circle and arc. I found that as long as the use cases are selected as "example_laser" and "example_laser_and_pen", it will happen when the simulated circle and arc have pop-up windows, and in the arc Another point of the phenomenon will stop. Other use cases do not have this problem.

"11111.png" is the use case I chose in this picture

[image]

svenhb, 2020-07-26:
Should be fixed in 1.5.0.1

lianzaozi, 2020-07-27:
Thank you very much.

The V1.5.0.1 version of the software has solved the problem of simulated arc pop-ups and pauses, and the problem of offset has also been solved.

svenhb, 2020-07-27:
Thank you for testing

closed 2020-07-27 by svenhb