svenhb/GRBL-Plotter #130: Suggest

lianzaozi, 2020-07-24:
I have a suggestion, can you make the width of the displayed path a configurable item, because if you display a very small geometric trajectory, the line displayed by the software is very thick, but sometimes the line is displayed when displaying a large geometric trajectory The width will be very thin. So I think it would be good to make the width of the display line a settable item.

lianzaozi, 2020-07-25:
Where are the fonts in the screenshot below stored?

[image]

svenhb, 2020-07-25:
These fonts are inside GCodeCreation\CreateFromForm\GCodeFromFontHershey.cs
Check here: https://gitlab.com/oskay/hershey-text/tree/master/hershey-text/svg_fonts
and here https://www.evilmadscientist.com/2019/hershey-text-v30/

lianzaozi, 2020-07-30:
Thank you, I know the source of these fonts.

What tools do you use to edit lff fonts? For example, I want to add Chinese to unicode.lff font, how do I do it? Do you mean that you can edit and make lff fonts using inkscape software?

Does the software after version V1.5.0.1 support the analysis of numbers and letters (ttf) in dxf files? I tested that the following numbers and letters can be displayed, but Chinese does not support display.

svenhb, 2020-07-30:
I didn't edit any font, I just took the data I found.  
With inkscape it may work: http://write.flossmanuals.net/start-with-inkscape/creating-fonts/#:~:text=Creating%20Custom%20Fonts,-With%20HTML5%20and&text=Fonts%20can%20be%20saved%20in,under%20Text%20%3E%20SVG%20Font%20Editor%20.

svenhb, 2020-10-17:
Can't help here.

closed 2020-10-17 by svenhb