svenhb/GRBL-Plotter #131: The plotter software opens gcode and reports an error

lianzaozi, 2020-07-25:
Open gcode with Chinese annotations with the software version between V1.3.4.4~V1.5.0.0, the software will pop up an error message, but use the version between V1.3.4.3~v1.3.3.0 When the software opens gcode with Chinese annotations, the software can be opened and displayed normally.

Test files
[DemoTest_UTF-16 BE.zip](https://github.com/svenhb/GRBL-Plotter/files/4975347/DemoTest_UTF-16.BE.zip)

Use V1.5.0.0 version software to open the error screenshot
[image]

svenhb, 2020-07-26:
Should be fixed in 1.5.0.1

lianzaozi, 2020-07-27:
Thank you very much, the Chinese comments are now displayed normally for the software version V1.5.0.1

closed 2020-07-27 by svenhb