svenhb/GRBL-Plotter #132: Increase the sorting number display, and mark the processing direction

lianzaozi, 2020-07-25:
The dxf2gcode software will add a sorting digital display for each track, and mark the processing direction. I think it is very good. Will future versions of the plotter software add types of functions? If you want to add, you can make the numbers indicating the sorting order and the arrows indicating the processing direction optional, so that the user can decide whether to show or hide.

The screenshot below is the display mode in dxf2gcode
[image]

lianzaozi, 2020-07-26:
The number displayed by dxf2gcode is in front of the layer, followed by the order of the primitives in the layer.

The image below is a screenshot of a dxf file with 4 layers.
[image]

Dxf file for testing
[shape.zip](https://github.com/svenhb/GRBL-Plotter/files/4976979/shape.zip)

svenhb, 2020-07-28:
Check new release for simple display of path direction and id in pen-up path

lianzaozi, 2020-07-29:
Some errors occurred in the software of version V1.5.0.2. I used the software of version V1.5.0.2 to open the test2.zip file and the dxf file obtained after decompression had an error. The traces of some primitives were not displayed. Use version V1 .5.0.1 software does not have this problem

The screenshot image1 below is a screenshot of the software version V1.5.0.2
[image]

image2 is a screenshot of the software version V1.5.0.1
[image]

test dxf file
[test2.zip](https://github.com/svenhb/GRBL-Plotter/files/4995434/test2.zip)

svenhb, 2020-07-29:
There was a bug in the implementation of #132 
Please check new release

lianzaozi, 2020-07-30:
The software version V1.5.0.3 fixed the problem that some geometric primitives in the software version V1.5.0.3 could not be displayed.

Now I have another question, that is, the arrow of the first point is not displayed. Is this due to other reasons?
[image]

Suggestion: Can you make three options on the setting page to show/hide arrows, show/hide numbers, and show/hide empty tracks. The user chooses which option to enable, and the corresponding item is displayed in the track display interface.

If the user enables the display/hide arrow and the display/hide number and the display/hide idling track function in the settings, the software interface displays idling track + arrow + direction, and the idling track is displayed and hidden at the same time when switching the space bar +arrow+direction.

If the user disables the display/hide arrow and disables the display/hide number function in the settings and enables the display of the idling track function at the same time, the software interface displays the idling track, and the idling track is displayed and hidden when the space bar is switched.

This will be very convenient for different users. Because when there are many geometric primitives, the idling track + arrow + direction are displayed at the same time, the software display interface will be messy. However, sometimes arrows and numbers are convenient for users to judge the trajectory sorting results and processing direction.

lianzaozi, 2020-07-30:
Is the reason why the arrow is not displayed on the first processing track, is it considered that sometimes the starting point coordinate of the first processing track is (0,0)?

svenhb, 2020-07-30:
Strange, with all graphics I tested, the first number and arrow will be displayed...
I tried with/without offset, or sorting...

lianzaozi, 2020-07-30:
This is the dxf file I used for testing
[test.zip](https://github.com/svenhb/GRBL-Plotter/files/4999818/test.zip)

svenhb, 2020-08-01:
First figure starts and ends on position 0;0 - hard to see:
[image]

lianzaozi, 2020-08-02:
You are right, the first number can be displayed, but the first arrow is not.

lianzaozi, 2020-08-02:
Hi svenhb
I found some patterns
1. The plotter software reads the svg file and can display the first number and arrow, but the plotter software reads the dxf file can display the first number but not the first arrow.

2. When the svg file is read for different times, the arrow and the number can sometimes be displayed correctly, and sometimes the two numbers 1 and the arrow are displayed.

The software version I tested is v1.5.0.3
The following are the files used for testing
[H.zip](https://github.com/svenhb/GRBL-Plotter/files/5011891/H.zip)

[line.zip](https://github.com/svenhb/GRBL-Plotter/files/5011892/line.zip)

svenhb, 2020-08-04:
Can be switched in Setup:
[image]

lianzaozi, 2020-08-05:
I am going to test it

lianzaozi, 2020-08-05:
The Gong that shows and hides arrows and numbers is very easy to use.
thank you

closed 2020-08-15 by svenhb