svenhb/GRBL-Plotter #133: Can it support RS-274D and RS-274X Gerber formats

raercan, 2020-07-25:
Hi , can u add RS-274D and RS-274X Gerber formats nc cut file formats for tangential fabric or vinyl cutter.  İts simple more than g code . 

Thank you .

svenhb, 2020-07-26:
Can you upload an example file?

raercan, 2020-07-27:
H1*M70*D2*M15*M20*LADIES-PANT/L=23.00cm/W=90.00c*N1*M26*M15*D2*M15*X0Y0*M14*X0Y196*X905Y196*X905Y0*X0Y0*M15*N2*M15*D2*M15*X0Y3346*M14*X905Y3346*X905Y3543*X0Y3543*X0Y3346*M15*QX905Y0*M0*
N,0001
L0*

raercan, 2020-07-27:
this cut file is 1 part . actually need to cut approximetly 10-100 parts . with conveyor. bite by bite. 

This machine is using tangential axis .like yours machine
https://www.youtube.com/watch?v=uMFyTCK8Ggw

svenhb, 2020-07-31:
Are you  sure your example code is correct gerber-code? I miss the '*' after each line.   
Also all the 'H' and 'M' commands are not described by gerber.

raercan, 2020-07-31:
`H1*M70*D2*M15*M20*LADIES-BLOUSE/L=23.00cm/W=90.00c*N1*M26*M15*D2*M15*X0Y0*M14*X0Y196*X905Y196*X905Y0*X0Y0*M15*N2*M15*D2*M15*X0Y3346*M14*X905Y3346*X905Y3543*X0Y3543*X0Y3346*M15*QX905Y0*M0*
N,0001
L0*
`

raercan, 2020-07-31:
sorry .github website is convert when i was paste codes

raercan, 2020-07-31:
and u can look this github "https://github.com/rsmith-nl/nctools"

svenhb, 2020-07-31:
Ok, good to know, but M-Code are still unknown:  
Example of known codes: https://www.artwork.com/gerber/gbr_rip/mcode_support.htm  
Also 'H1' and 'N1' I can't find...

svenhb, 2020-07-31:
>and u can look this github "https://github.com/rsmith-nl/nctools"   
  
Does it work with your example-code?

raercan, 2020-07-31:
this  program is dxf to gerber cutter cut file converter  for cutter machine.  may u can look in  how is work gerber codes.

raercan, 2020-07-31:
https://www.youtube.com/watch?v=8nt_5XFGXHU

svenhb, 2020-08-04:
May work in 1.5.0.4

raercan, 2020-08-08:
can you improve about notch ?  can you add notch processor for I notch change with V notch .  when program is look I notch on pattern it will be change with V shape notch.  Should be V notch set deep lengh and dont pen up pen down  when cutting v notch. it is possible ? 
Viele Grüß

raercan, 2020-08-08:
`H1*M70*M20*TRIBUNE 5 CEP BAYAN*ZX4618*N1*D2*M15*X1490Y745*D1*X1490Y687*D2*X1471Y745*D1*X1510Y745*D2*X1523Y687*D1*X1523Y745*X1548Y745*X1561Y738*X1561Y721*X1555Y715*X1523Y715*D2*X1544Y715*D1*X1562Y687*D2*X1579Y745*D1*X1579Y687*D2*X1593Y687*D1*X1593Y745*X1614Y745*X1623Y736*X1623Y727*X1597Y718*X1622Y718*X1632Y707*X1632Y696*X1623Y687*X1593Y687*D2*X1648Y745*D1*X1648Y696*X1657Y687*X1678Y687*X1687Y696*X1687Y745*D2*X1703Y687*D1*X1703Y745*X1742Y687*X1742Y745*D2*X1792Y687*D1*X1756Y687*X1756Y745*X1792Y745*D2*X1756Y718*D1*X1778Y718*D2*X1837Y694*D1*X1845Y687*X1863Y687*X1873Y697*X1873Y715*X1862Y726*X1844Y726*X1844Y745*X1873Y745*D2*X1957Y736*D1*X1948Y745*X1925Y745*X1916Y736*X1916Y696*X1925Y687*X1948Y687*X1957Y696*D2*X2007Y687*D1*X1971Y687*X1971Y745*X2007Y745*D2*X1971Y718*D1*X1993Y718*D2*X2021Y687*D1*X2021Y745*X2046Y745*X2059Y738*X2059Y721*X2052Y715*X2021Y715*D2*X2102Y687*D1*X2102Y745*X2123Y745*X2132Y736*X2132Y727*X2107Y718*X2132Y718*X2142Y707*X2142Y696*X2132Y687*X2102Y687*D2*X2149Y687*D1*X2171Y745*X2193Y687*D2*X2157Y706*D1*X2185Y706*D2*X2202Y745*D1*X2224Y718*X2224Y687*D2*X2246Y745*D1*X2224Y718*D2*X2254Y687*D1*X2276Y745*X2299Y687*D2*X2263Y706*D1*X2290Y706*D2*X2315Y687*D1*X2315Y745*X2354Y687*X2354Y745*D2*X1912Y569*D1*X1902Y588*X1902Y608*X1912Y627*X1931Y627*X1941Y608*X1941Y588*X1931Y569*X1912Y569*D2*X1957Y569*D1*X1957Y627*X1996Y569*X1996Y627*D2*X2038Y569*D1*X2038Y627*X2063Y627*X2076Y620*X2076Y603*X2070Y597*X2038Y597*D2*X2085Y569*D1*X2107Y627*X2130Y569*D2*X2094Y588*D1*X2121Y588*D2*X2185Y618*D1*X2176Y627*X2153Y627*X2143Y618*X2143Y578*X2153Y569*X2176Y569*X2185Y578*D2*X2193Y569*D1*X2215Y627*X2237Y569*D2*X2201Y588*D1*X2229Y588*D2*X2082Y509*D1*X2118Y509*X2099Y491*X2109Y491*X2118Y482*X2118Y461*X2107Y451*X2088Y451*X2079Y460*D2*X2143Y491*D1*X2134Y481*X2134Y460*X2143Y451*X2161Y451*X2170Y460*X2170Y481*X2161Y491*X2143Y491*X2134Y500*X2143Y509*X2161Y509*X2170Y500*X2161Y491*D2*X2052Y380*D1*X2065Y391*X2065Y333*D2*X2108Y391*D1*X2108Y333*X2138Y333*D2*X3190Y623*D1*X3387Y544*X2418Y544*X1449Y544*X885Y544*D2*X786Y53*D1*X792Y1004*D2*X567Y885*D1*X562Y46*D2*X60Y844*D1*X573Y917*X662Y929*X669Y931*X693Y937*X701Y939*X711Y943*X723Y949*X730Y952*X737Y957*X749Y965*X754Y969*X762Y977*X770Y987*X775Y994*X780Y1002*X785Y1014*X789Y1025*X794Y1051*X1022Y996*X1043Y992*X1229Y966*X1394Y951*X1790Y927*X1871Y925*X2233Y926*X3202Y953*X4170Y981*X4520Y991*X4518Y107*X3550Y118*X2581Y130*X2256Y134*X1972Y127*X1908Y125*X1701Y112*X1510Y98*X1059Y69*X824Y53*X791Y52*X634Y47*X561Y47*X437Y51*X374Y55*X268Y65*X277Y180*X280Y283*X280Y300*X279Y320*X276Y348*X273Y362*X271Y372*X268Y386*X264Y396*X261Y403*X257Y413*X252Y422*X246Y431*X238Y440*X230Y448*X219Y457*X210Y464*X195Y473*X178Y481*X172Y484*X156Y490*X150Y492*X136Y495*X124Y498*X107Y500*X73Y504*X49Y506*X58Y708*X60Y844*D2*X13Y885*M14*X521Y957*X522Y946*M15*X521Y957*M14*X567Y964*X653Y975*X679Y982*X687Y985*X701Y991*X710Y996*X716Y1000*X721Y1005*X728Y1011*X732Y1018*X738Y1027*X741Y1034*X747Y1047*X749Y1055*X752Y1063*X755Y1079*X762Y1119*X1033Y1054*X1057Y1049*X1214Y1027*X1398Y1010*X1799Y986*X1873Y984*X2233Y985*X2233Y973*M15*X2233Y985*M14*X3201Y1012*X4169Y1040*X4520Y1050*X4520Y1038*M15*X4520Y1050*M14*X4618Y1047*X4617Y78*X4617Y60*X4518Y59*X4519Y71*M15*X4518Y59*M14*X3550Y71*X2581Y83*X2257Y86*X2257Y98*M15*X2257Y86*M14*X1966Y80*X1910Y77*X1712Y65*X1513Y51*X1062Y22*X839Y7*X793Y4*X635Y0*X560Y0*X446Y4*X371Y8*X225Y21*X238Y181*X241Y269*X241Y286*X239Y317*X238Y334*X236Y346*X234Y358*X232Y367*X228Y380*X223Y392*X219Y399*X214Y406*X206Y415*X199Y422*X193Y427*X186Y432*X175Y438*X160Y445*X151Y449*X138Y453*X127Y457*X107Y461*X90Y464*X55Y468*X0Y472*X11Y696*X11Y709*X13Y885*M15*M0*`

svenhb, 2020-08-08:
I don't understand your text, what do you mean with I- and V-notch?
I assume you may talk about tangential knife lifting up on sharpe curves?
[image]

raercan, 2020-08-08:
#[https://hizliresim.com/uzXKQs](url)

raercan, 2020-08-08:
[https://hizliresim.com/uzXKQs](url)

raercan, 2020-08-08:
https://hizliresim.com/uzXKQs

raercan, 2020-08-08:
you can see 2 between 2  i mean small vertical lines on pattern borders.

svenhb, 2020-08-08:
Do you have a screenshot how it should look like? I don't understand the mix of D1/D2 Pen up/down and M14/M15 Knife up/down.
[image]
Doesn't look correct  
[image]

raercan, 2020-08-08:
`H1*ZX19000*M70*N1*M26*D2*M15*X5379Y2965*M14*X5457Y2903*X5517Y2901*X5627Y2901*X5737Y2903*X5847Y2909*X5947Y2918*X6065Y2931*X6182Y2947*X6299Y2965*X6329Y2972*X6454Y2998*X6553Y3022*X6649Y3050*X6721Y2865*X6721Y2865*M19*X6668Y2847*D2*M15*X6721Y2865*M14*X7165Y1744*X7165Y1744*M19*X7112Y1726*D2*M15*X7165Y1744*M14*X7251Y1485*X7158Y1470*X6880Y1407*X6849Y1401*X6769Y1382*X6740Y1374*X6668Y1350*X6473Y749*X6473Y749*M19*X6420Y767*D2*M15*X6473Y749*M14*X6498Y631*X6526Y495*X6553Y359*X6569Y267*X6592Y134*X6614Y0*X6570Y3*X6428Y13*X6286Y25*X6144Y37*X6002Y50*X5860Y64*X5718Y78*X5651Y86*X5511Y102*X5371Y120*X5231Y138*X5091Y158*X4951Y179*X4900Y187*X4778Y208*X4656Y231*X4535Y257*X4443Y279*X4318Y311*X4193Y346*X4069Y382*X4069Y382*M19*X4069Y438*D2*M15*X4069Y382*M14*X2391Y382*X650Y383*X628Y385*X559Y392*X491Y404*X424Y421*X398Y429*X353Y447*X310Y469*X269Y494*X257Y504*X235Y526*X216Y550*X200Y576*X187Y603*X185Y608*X170Y651*X159Y695*X153Y739*X150Y783*X0Y783*X0Y981*X45Y2113*X51Y2213*X299Y2203*X299Y2203*M19*X297Y2147*D2*M15*X299Y2203*M14*X449Y2196*X449Y2196*M19*X447Y2140*D2*M15*X449Y2196*M14*X883Y2179*X883Y2179*M19*X881Y2123*D2*M15*X883Y2179*M14*X939Y2175*X1079Y2164*X1219Y2153*X1359Y2140*X1499Y2126*X1532Y2123*X1677Y2107*X1822Y2091*X1967Y2075*X2112Y2058*X2257Y2040*X2402Y2022*X2402Y2022*M19*X2395Y1966*D2*M15*X2402Y2022*M14*X2444Y2021*X2552Y2017*X2660Y2017*X2705Y2019*X2838Y2023*X2971Y2030*X3104Y2037*X3121Y2039*X3260Y2049*X3399Y2061*X3538Y2074*X3677Y2088*X3751Y2097*X3868Y2113*X3984Y2132*X3984Y2132*M19*X3995Y2077*D2*M15*X3984Y2132*M14*X4021Y2143*X4124Y2175*X4151Y2184*X4242Y2218*X4242Y2218*M19*X4257Y2164*D2*M15*X4242Y2218*M14*X4336Y2247*X4432Y2282*X4527Y2321*X4627Y2368*X4723Y2418*X4817Y2472*X4823Y2476*X4934Y2558*X5042Y2644*X5147Y2733*X5198Y2779*X5291Y2870*X5379Y2965*D2*M15*N2*M25*D2*M15*X86Y3359*M14*X331Y3351*X335Y3350*X335Y3350*M19*X330Y3294*D2*M15*X335Y3350*M14*X483Y3346*X486Y3345*X486Y3345*M19*X481Y3289*D2*M15*X486Y3345*M14*X617Y3336*X754Y3325*X891Y3312*X1028Y3298*X1134Y3286*X1270Y3269*X1406Y3251*X1541Y3231*X1635Y3218*X1774Y3199*X1913Y3180*X2053Y3163*X2108Y3157*X2234Y3145*X2360Y3135*X2360Y3135*M19*X2361Y3079*D2*M15*X2360Y3135*M14*X2478Y3146*X2595Y3159*X2600Y3160*X2719Y3178*X2838Y3198*X2867Y3204*X3009Y3231*X3151Y3259*X3292Y3288*X3409Y3313*X3548Y3344*X3687Y3376*X3692Y3377*X3692Y3377*M19*X3702Y3322*D2*M15*X3692Y3377*M14*X3893Y3424*X3893Y3424*M19*X3907Y3370*D2*M15*X3893Y3424*M14*X3951Y3441*X4070Y3479*X4167Y3513*X4295Y3559*X4372Y3588*X4415Y3517*X4427Y3498*X4444Y3476*X4465Y3453*X4471Y3448*X4486Y3436*X4502Y3427*X4519Y3419*X4522Y3418*X4599Y3395*X4606Y3394*X4671Y3380*X4675Y3379*X4675Y3379*M19*X4668Y3323*D2*M15*X4675Y3379*M14*X4740Y3379*X4815Y3380*X4826Y3382*X4826Y3382*M19*X4828Y3326*D2*M15*X4826Y3382*M14*X5026Y3437*X5116Y3460*X5204Y3413*X5139Y3292*X5127Y3269*X5074Y3170*X5062Y3149*X5002Y3049*X4971Y2997*X4905Y2890*X4871Y2839*X4799Y2735*X4738Y2657*X4676Y2584*X4656Y2562*X4572Y2475*X4483Y2393*X4450Y2366*X4373Y2305*X4293Y2251*X4293Y2251*M19*X4269Y2302*D2*M15*X4293Y2251*M14*X4286Y2247*X4216Y2214*X4144Y2185*X4126Y2179*X4057Y2158*X3994Y2143*X3994Y2143*M19*X3989Y2199*D2*M15*X3994Y2143*M14*X3987Y2142*X3926Y2140*X3820Y2140*X3714Y2143*X3684Y2145*X3557Y2153*X3430Y2164*X3304Y2177*X3208Y2187*X3075Y2199*X2942Y2210*X2809Y2218*X2754Y2221*X2625Y2226*X2496Y2228*X2367Y2228*X2364Y2227*X2364Y2227*M19*X2364Y2283*D2*M15*X2364Y2227*M14*X2242Y2220*X2102Y2211*X1962Y2204*X1822Y2198*X1682Y2192*X1542Y2188*X1516Y2188*X1380Y2185*X1244Y2185*X1108Y2186*X972Y2188*X836Y2191*X836Y2191*M19*X839Y2247*D2*M15*X836Y2191*M14*X402Y2208*X402Y2208*M19*X405Y2264*D2*M15*X402Y2208*M14*X253Y2214*X249Y2214*X249Y2214*M19*X255Y2270*D2*M15*X249Y2214*M14*X0Y2224*X8Y2332*X79Y3270*X86Y3359*D2*M15*N3*M26*D2*M15*X5379Y3236*M14*X5291Y3331*X5198Y3422*X5147Y3468*X5042Y3557*X4934Y3643*X4823Y3725*X4817Y3729*X4723Y3783*X4627Y3833*X4527Y3880*X4432Y3919*X4336Y3954*X4242Y3983*X4242Y3983*M19*X4257Y4037*D2*M15*X4242Y3983*M14*X4151Y4017*X4124Y4026*X4021Y4058*X3984Y4069*X3984Y4069*M19*X3995Y4124*D2*M15*X3984Y4069*M14*X3868Y4088*X3751Y4104*X3677Y4113*X3538Y4127*X3399Y4140*X3260Y4152*X3121Y4162*X3104Y4164*X2971Y4171*X2838Y4178*X2705Y4182*X2660Y4184*X2552Y4184*X2444Y4180*X2402Y4179*X2402Y4179*M19*X2395Y4235*D2*M15*X2402Y4179*M14*X2257Y4161*X2112Y4143*X1967Y4126*X1822Y4110*X1677Y4094*X1532Y4078*X1499Y4075*X1359Y4061*X1219Y4048*X1079Y4037*X939Y4026*X883Y4022*X883Y4022*M19*X881Y4078*D2*M15*X883Y4022*M14*X449Y4005*X449Y4005*M19*X447Y4061*D2*M15*X449Y4005*M14*X299Y3998*X299Y3998*M19*X297Y4054*D2*M15*X299Y3998*M14*X51Y3988*X45Y4088*X0Y5220*X0Y5418*X150Y5418*X153Y5462*X159Y5506*X170Y5550*X185Y5593*X187Y5598*X200Y5625*X216Y5651*X235Y5675*X257Y5697*X269Y5707*X310Y5732*X353Y5754*X398Y5772*X424Y5780*X491Y5797*X559Y5809*X628Y5816*X650Y5818*X2391Y5819*X4069Y5819*X4069Y5819*M19*X4069Y5763*D2*M15*X4069Y5819*M14*X4193Y5855*X4318Y5890*X4443Y5922*X4535Y5944*X4656Y5970*X4778Y5993*X4900Y6014*X4951Y6022*X5091Y6043*X5231Y6063*X5371Y6081*X5511Y6099*X5651Y6115*X5718Y6123*X5860Y6137*X6002Y6151*X6144Y6164*X6286Y6176*X6428Y6188*X6570Y6198*X6614Y6201*X6592Y6067*X6569Y5934*X6553Y5842*X6526Y5706*X6498Y5570*X6473Y5452*X6473Y5452*M19*X6420Y5434*D2*M15*X6473Y5452*M14*X6668Y4851*X6740Y4827*X6769Y4819*X6849Y4800*X6880Y4794*X7158Y4731*X7251Y4716*X7165Y4457*X7165Y4457*M19*X7112Y4475*D2*M15*X7165Y4457*M14*X6721Y3336*X6721Y3336*M19*X6668Y3354*D2*M15*X6721Y3336*M14*X6649Y3151*X6553Y3179*X6454Y3203*X6329Y3229*X6299Y3236*X6182Y3254*X6065Y3270*X5947Y3283*X5847Y3292*X5737Y3298*X5627Y3300*X5517Y3300*X5457Y3298*X5379Y3236*D2*M15*N4*M26*D2*M15*X0Y6094*M14*X20Y6193*X20Y6195* [... cut ...] 7156Y11906*D2*M15*X17100Y11907*M14*X17101Y12104*X17101Y12104*M19*X17157Y12104*D2*M15*X17101Y12104*M14*X17101Y12303*X17101Y12303*M19*X17157Y12303*D2*M15*X17101Y12303*M14*X17100Y12427*X17095Y12551*X17088Y12675*X17081Y12771*X17069Y12871*X17054Y12971*X17033Y13072*X17033Y13072*M19*X17088Y13084*D2*M15*X17033Y13072*M14*X17010Y13154*X16983Y13234*X16967Y13274*X16909Y13412*X16849Y13550*X16789Y13687*X16727Y13823*X16693Y13896*X16669Y13947*X16700Y13994*X17093Y14587*X17116Y14622*D2*M15*D4*X18180Y0*D2*M15*M0*`

raercan, 2020-08-08:
https://youtu.be/xylEcHtj1OM?t=435

if u can watch this video . he is give us notch example and explaining.

can we change  notch shape ? maybe u can add this control.

raercan, 2020-08-08:
and i want to use pnumatic cylinder for Z axis . how can i do that ?  its could be faster movement for cutting. no need precision movement for Z axis when we want to use tangential axis movement for A axis.

raercan, 2020-08-08:
https://www.youtube.com/watch?v=wZPcUahD-SQ

like this  machine.  

machine is starting for cut . pattern by pattern is cutting.   this machine is using air cylinder for z axis up/down movement. 
and used pattern number every parts.   
when start is cutting  z axis is down . when  ending z axis is up . cause  its using V notch .
 if u use I notch  blade is need to up.  cause A axis is not turning with blade . V nocth is have to for good quality cutting.

svenhb, 2020-08-08:
The notch shape is already given in your code... No chance for you  to create your code with V-shape notch?  
For Z-Axis you may use spindel on/off signal to switch a valve for your pneumatic cylinder.  
[image]

raercan, 2020-08-08:
yes . its possible in pattern software .  

and thank you for spindle on/off function  explaining.

can u do show on part number of patterns. and some pattern is looking like exploded.  some pattern is cutting multiparts or halfparts. 

best regards

svenhb, 2020-08-15:
Now output looks right to me.
[image]

raercan, 2020-08-15:
will machine is cut outside border ?

raercan, 2020-08-15:
can we seperate  for layers ?

raercan, 2020-08-15:
can we select  starting corner or point ?

raercan, 2020-08-15:
can we see part number ? and can we select what we want starting part  ? or delete /skip / part or parts ?

svenhb, 2020-08-15:
can we seperate for layers ? 
Enable Grouping on Layers:
[image]
Select group via Ctrl-Click in 2D-view:
[image]

raercan, 2020-08-15:
i cant see my  files in there can u share with me this  file .

svenhb, 2020-08-15:
can we select starting corner or point ? no, because these are not closed path as you can see.

svenhb, 2020-08-15:
can we see part number ? yes 
and can we select what we want starting part ? no 
or delete /skip / part or parts ? yes  
we need to contiune after emergency stop or intterrupt . please support this sitution. 
It is implemented - I need to write wiki....

raercan, 2020-08-15:
[https://hizliresim.com/MCD0dp](url)

can u add  in right-click sub menu (" sort by "layer=sequence " )

svenhb, 2020-08-15:
Can be done during import in my previous example "revert Order" was checked, without reversing it should be ok
[image]

raercan, 2020-08-15:
and we want to use  blade for cutting. sometime corners is not  rigth cutting. we need to blade compensation. i mean blade is starting point  for little bit back  and out little bit forward .

raercan, 2020-08-15:
i want to send to you cutter user manual .  how can i mail ? i cant find your adress.

raercan, 2020-08-15:
is it possible like this conveyorized ? 

https://www.youtube.com/watch?v=JMLyodv1nT0

svenhb, 2020-08-17:
we need to blade compensation.:  
https://github.com/svenhb/GRBL-Plotter/wiki/Drag-tool-compensation  
or 
https://github.com/svenhb/GRBL-Plotter/wiki/Import-options#tangential-axis-commands  
 
is it possible like this conveyorized ? or part by part or bite by bite ? 
no

svenhb, 2020-10-17:
I think import is possible

closed 2020-10-17 by svenhb