svenhb/GRBL-Plotter #134: Regarding the problem that the mouse in the gcode editing area cannot select multiple lines and the middle layer name in the dxf file is garbled

lianzaozi, 2020-07-26:
FastColoredTextBox.dll supports pressing the left mouse button to select multiple lines. But in the plotter software, press the left mouse button and drag, but nothing is selected.

[image]

FastColoredTextBox.dll supports Chinese display, but the Chinese layer name in dxf will be displayed as garbled characters. The dxf path in the screenshot image5 has Chinese in the dxf path, which can be displayed normally, but the Chinese layer name is displayed as garbled characters.
[image]

Another software that uses FastColoredTextBox.dll can hold down the left mouse button and drag the mouse to select multiple items. And it also supports pressing the shift key to select the content between the two mouse clicks. This function should be provided by FastColoredTextBox.dll.

123.gif animation is another software that uses FastColoredTextBox.dll
[image]

svenhb, 2020-07-26:
Cab you upload this file for testing?
How did you open your file?

lianzaozi, 2020-07-27:
Sorry I forgot to upload my test file. I used several different dxf files to test and found that the Chinese layer name in test1 was garbled, and the Chinese layer name in test2 was displayed normally. I use CAD to open the test1 and test2 files in the cad software, and the Chinese layer names of these two files can be displayed normally.

Chinese layer names display garbled characters
[test1.zip](https://github.com/svenhb/GRBL-Plotter/files/4979132/test1.zip)

Chinese layer names are displayed normally
[test2.zip](https://github.com/svenhb/GRBL-Plotter/files/4979135/test2.zip)

lianzaozi, 2020-07-27:
The way I open the file:
  "File"=>"open file" Select the .nc file to be opened, and click the ok button.

svenhb, 2020-07-28:
If I open both files in LibreCad, test2 looks normal, test1 is already garbled - so the source is not ok:
[image]

lianzaozi, 2020-07-28:
Thank you, I just used librecad to test the layer name of the test1 file is garbled. I used another brand of cad software to open the test1 file. In that cad software, the layer names of test1 and test2 can be displayed normally.

I will only use librecad when testing in the future, which can avoid some strange problems.

svenhb, 2020-07-28:
>I will only use librecad when testing in the future, which can avoid some strange problems.   
  
Good idea, thanks

closed 2020-07-28 by svenhb