svenhb/GRBL-Plotter #135: The clockwise arc generated is not G02 but G0 (I have turned off the function of sorting by the shortest path), and the x coordinate of the generated arc is incorrect.

lianzaozi, 2020-07-26:
I found a strange phenomenon
I only drew an arc 1 in the dxf file (direction: counterclockwise, center angle 270 degrees, start point coordinates (50, 0), center coordinates (50, 50), end point coordinates (0, 50))
with
Arc 2 (Clockwise _ central angle 270 degrees _ Starting point (50, 100) _ Circle center (150, 100) _ end point (100, 50)).

When I open the dxf file containing arc 1 with plotter software, the generated GCode is
G03 X-0.190 Y50.000 I0.000 J50.000 F800
Why is the X coordinate not X0.000 but X-0.190? When testing, I canceled sorting by the shortest path in the settings of the plotter software.

The GCode generated by dxf2gcode is as follows. I think the coordinates generated by dxf2Gcode are correct.
G03 X 0.000 Y 50.000 I 0.000 J 50.000

test dxf file
[arc1  Counterclockwise_Central angle 270 degrees_Start point (0,50)_Circle center (50,50)_End point (50,0).zip](https://github.com/svenhb/GRBL-Plotter/files/4977779/arc1.Counterclockwise_Central.angle.270.degrees_Start.point.0.50._Circle.center.50.50._End.point.50.0.zip)

When I use the plotter software to open the dxf file containing arc 2 the GCode generated is
G03 X50.000 Y100.000 I0.000 J50.000 F800
Shouldn't the clockwise arc generate G02? During the test, in the settings of the plotter software, I canceled sorting by the shortest path and canceled the offset to (0,0) point.

test dxf file

[arc2  Clockwise _ central angle 270 degrees _ Starting point (50, 100) _ Circle center (150, 100) _ end point (100, 50).zip](https://github.com/svenhb/GRBL-Plotter/files/4977780/arc2.Clockwise._.central.angle.270.degrees._.Starting.point.50.100._.Circle.center.150.100._.end.point.100.50.zip)

svenhb, 2020-07-26:
There is an offset (I don't know why), when removing the offset, it is ok.
[image]

Offset removed:  
[image]

svenhb, 2020-07-26:
Should be fixed in 1.5.0.1

lianzaozi, 2020-07-27:
Cancel the use of the "Sort objects by shortest connections" function. Does the plotter software sort according to the drawing order of dxf? arc2 I draw in a clockwise direction. If it is a plotter software, it generates GCode according to my drawing direction. Then this GCode should generate G02, but the current GCode generated by the plotter software is G03.

svenhb, 2020-07-27:
DXF arcs are always counter clockwise - so I can't fix it:
https://ezdxf.readthedocs.io/en/stable/dxfentities/arc.html
https://forums.autodesk.com/t5/autocad-forum/how-to-find-whether-an-arc-is-clockwise-or-counter-clockwise-in/td-p/7657875

lianzaozi, 2020-07-27:
Thank you still very much, now the coordinate values of the GCode generated by the arc are correct.

lianzaozi, 2020-07-27:
Can Bugle determine whether the arc is clockwise or counterclockwise?

Bugle is a positive number, the arc is counterclockwise; a negative number is clockwise

svenhb, 2020-07-27:
Thank you for testing

closed 2020-07-27 by svenhb