svenhb/GRBL-Plotter #137: Do not understand the use of these functions?

lianzaozi, 2020-07-29:
The red parts 1, 2, 3, and 4 in the screenshot below are not clear. Can you explain the effect in detail?
[image]

svenhb, 2020-07-29:
1) and 2) if you want to scale both axis to a specific end value for x or y use these options (it also works with G2 / G3). E.g. you want a width of exactly 100 mm, use 1)   

3) and 4) only scales the given axis and distorts the graphics... It does not work with G2 / G3 commands

lianzaozi, 2020-07-30:
Thank you very much for your reply, I understand.

lianzaozi, 2020-07-30:
I don't quite understand the role of the options in the screenshot below.
[image]

svenhb, 2020-07-30:
There is an option "Rotary axis control instead of X or Y", intended e.g. for Egg-bot:
1) To draw the substituted axis without distortion I need the diameter to calculate the angle
2) / 3) If you want to draw on a half cylinder circumfence you put in here 180° 
  
Regarding Router / Laser:  
4) cutter radius compensation: moves the path a bit to the left or right side of the path to compensate the tool width of a router   
5) S controls the laser-power on some systems, a user asked to convert Z value to Laser power....

lianzaozi, 2020-07-31:
Thank you very much, I want to study it carefully.

lianzaozi, 2020-08-12:
What is the function of the two options in the red box in the screenshot below? Which use cases do these two functions work for?

[image]

svenhb, 2020-08-12:
"Compress G-Code" tries to avoid repetition of G1 commands (it's a modal command which needs to send once, not every line). Also the commands of non-changing axes values. - to save data size but it is harder to read.

"Relative movements" will use G91 relative coordinates and therfore recalculate the X and Y values. - good to create code for subroutines: you move to an absolute position, then switch to relative coordinates and run your subroutine.  
Example gcode_subroutine:
[image]

lianzaozi, 2020-08-13:
thank you for your reply.

After testing, I have some questions, I hope you can help me answer them.

About the scope of application of GCode Transform function?

GCode rotation, GCode scaling, GCode mirroring in the menu bar "G-Code Transform"... Do these functions only work in the absolute coordinate system? I found that when I generate the GCode of the relative coordinate system (G91) and then perform the GCode rotation, scaling, and mirroring functions, the path will be transformed into a path that is completely different from the original path. Perform GCode rotation, zoom, and mirror functions under the absolute coordinate system. These functions will work very well.

svenhb, 2020-08-13:
When implementing Scaling, Rotation and Mirroring I never tested with G91.

svenhb, 2020-08-15:
I fixed this: 
Scaling, Rotation and Mirroring on G-Code with G91.

lianzaozi, 2020-08-17:
Thank you very much, I will do some tests.

lianzaozi, 2020-08-17:
Hi svenhb, I don’t quite understand what 1, 2, and 3 in the screenshot below mean and how?
[image]

svenhb, 2020-08-18:
1) will fix the 1st view to the machine limits, which needs to be setup:
[image]

svenhb, 2020-08-18:
2. Tool table shows the tool exchange positions:
[image]

svenhb, 2020-08-18:
3. shows the gray background figures - e.g. will be set before GCode translation:
After rotation of the rectangle, the original position is still visible via the background path:
[image]

closed 2020-10-17 by svenhb