svenhb/GRBL-Plotter #138: SVG file opening error, the same file can be normal in the previous version

lianzaozi, 2020-08-02:
The following SVG file can be opened normally with software before version V1.3.4.6. The software between version V1.5.0.0 and version V1.5.0.3 cannot open this file.

Test SVG file
[SVG_Test1.zip](https://github.com/svenhb/GRBL-Plotter/files/5011558/SVG_Test1.zip)

image1 is the screenshot I opened with the software version V1.3.4.6
[image]

image2 is the screenshot I opened with the software version V1.5.0.3
[image]

svenhb, 2020-08-02:
I could fix it, but I think this SVG-code is not good, just one graphic-line per code-line - so much overhead to draw a line...
```
<g clip-path="url(#cp0)">
<path transform="matrix(.12,0,0,-.12,0,3061)" stroke-width="8" stroke-linecap="round" stroke-linejoin="round" fill="none" stroke="rgb(0,0,0)" d="M 5588.992 250.07843 L 39769.008 250.07843 "/>
<path transform="matrix(.12,0,0,-.12,0,3061)" stroke-width="8" stroke-linecap="round" stroke-linejoin="round" fill="none" stroke="rgb(0,0,0)" d="M 39769.008 250.07843 L 39769.008 25257.922 "/>
```

lianzaozi, 2020-08-04:
Thank you for your reply, this svg file is converted from dxf file using converter software.

lianzaozi, 2020-08-04:
I open the svg file that comes with the plotter software (path GRBL-Plotter-1.5.0.3\GRBL-Plotter\data\examples). The plotter software will also report an error. There is no problem to open it with the plotter software version before V1.3.4.6.

My operation process is recorded in git animation

[image]

svenhb, 2020-08-04:
Should be fixed in 1.5.0.4

lianzaozi, 2020-08-05:
The plotter software version v1.5.0.4 opens the svg file and it works normally.

svenhb, 2020-08-05:
Thanks for feedback

closed 2020-08-05 by svenhb