svenhb/GRBL-Plotter #139: Switch processing direction

lianzaozi, 2020-08-02:
Can you add the function of switching direction to the plotter software? This function is very easy to use in dxf2gcode. In the laser processing industry, the processing starting point of some paths is fixed, and sometimes the starting point sorted by the plotter software does not meet the process requirements. Just like the operation of the dxf2gcode software below, when I select a certain geometric primitive, it allows me to switch the processing direction.

gif of dxf2gcode software
[image]

As shown below, image1 is a sort without switching the processing direction.
[image]

image2 is the sort after the user manually switches the processing direction. (After the direction is switched, even if the user manually adjusts the sort order between the primitives, the start and end points of the primitives in the direction of the switch will remain in the state after the user switched)
[image]

lianzaozi, 2020-08-04:
For open-loop geometric shapes, there are start points and end points. Switching direction is generally to switch start and end points

If it is a closed graph, can the user select a certain point on the graph as the starting point? For example, closed geometric circles, closed geometric ellipses, closed geometric polygons, closed geometric rectangles, irregular shapes with closed ends...

svenhb, 2020-08-04:
Try new option in new release:  
[image]

lianzaozi, 2020-08-05:
After switching the direction of the primitive, some problems occurred. After the switch, the path of some geometric primitives would be repeated, and some would not be repeated, and the display of arrows and numbers was also confused.

Test dxf file
[2007.zip](https://github.com/svenhb/GRBL-Plotter/files/5025824/2007.zip)

[image]

svenhb, 2020-08-15:
Should be fixed now

lianzaozi, 2020-08-16:
The normal processing operation flow is like this:
1. Import vector graphics, the software automatically sorts the paths according to the user's set software sorting method
2. Then the user will check whether the processing sequence of each path meets the requirements one by one based on the processing sequence after the software sorting according to the processing requirements. Some users who do not meet the requirements will manually adjust the sequence (the mouse can right-click in the track display area at this time Press the mouse button and select a function "Sort according to user process and reverse processing direction" in the pop-up menu. When the user enables this function, the plotter software should close some built-in sorting algorithms at this time, for example: close sorting by distance.. ....) Then the user can manually sort according to his own process requirements and reverse the processing direction of some trajectories. These changes are only effective for the current trajectory selected by the user. Reversing the processing direction should not affect the sorting result. The current processing method of the plotter software: If I manually sort, and then reverse the direction of some primitives, the sort result is changed by the function of reversing the trajectory direction.

3. The correct approach is: the user manually adjusts the processing sequence without affecting the result of the processing direction. The user's manual adjustment of the processing direction will not affect the processing sequence.

As you can see in the gif animation I made, I first manually adjusted the processing sequence. After the adjustment, the processing sequence of the ellipse arc was 3. Then I reversed the processing direction of the ellipse arc. However, after the reversal, its processing order becomes 7 again. The correct way is that the processing order of the elliptical arc is still 3 after the reversal. I feel that after I reverse the processing direction of the elliptical arc, the plotter software sorts according to a certain sorting method, and this sorting is disrupted The processing sequence adjusted by the user is added.
[image]

svenhb, 2020-10-17:
I think I fixed it in 1.5.2.0

lianzaozi, 2020-10-19:
Thank you very much for your hard work. I will run some tests.

lianzaozi, 2020-10-28:
The function of switching the processing direction of the plotter software version 1.5.2.2 does not work properly.

lianzaozi, 2020-11-04:
When I was testing to switch the processing direction of the reversed elliptical arc, the software actually reversed the processing direction of the next path, and did not switch the processing direction of the elliptical arc.

Test steps:
First open the dxf file, then adjust the processing order of the elliptical arc to 3, and execute the reverse elliptical arc processing direction. But the software reverses the processing direction of the straight line with the processing order 4.

[image]

Test dxf file
[2008switch direction.zip](https://github.com/svenhb/GRBL-Plotter/files/5485366/2008switch.direction.zip)

svenhb, 2020-11-04:
Should be fixed with this version: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1523_publish.zip

lianzaozi, 2020-11-09:
I was unable to download version 1.5.3.3 from the url above. Failed to download correctly when downloading.

svenhb, 2020-11-09:
Too bad, I just tried and had no problems. https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1523_publish.zip

lianzaozi, 2020-11-10:
Hello, I tested the software version 1.5.3.3 and found two problems that I don’t understand

1. When the software reverses the processing direction of the spline curve, the software does not reverse the correct direction, but keeps the original processing direction
[image]

2. When I choose to rotate the G code by 180, some trajectories will be displayed in gray in the place where the trajectory was originally displayed after the reversal. I don't understand what these gray tracks do.
[image]

Test dxf
[2008switch direction.zip](https://github.com/svenhb/GRBL-Plotter/files/5514209/2008switch.direction.zip)

GIF
[image]

svenhb, 2020-11-10:
It is version 1.5.2.3 (not 1.5.3.3)
In your example any sorting is switched off, the spline (figure 13) is the last object - any need to reverse?
The grey background objects shows the previous position, which can may be restored via 'undo'

lianzaozi, 2020-11-11:
You're right, it's version 1.5.2.3 (not 1.5.3.3).

lianzaozi, 2020-11-13:
In some cases, the last geometry element may also need to reverse the machining trajectory.

svenhb, 2020-11-21:
Now the last object should also change.
Check new release: https://github.com/svenhb/GRBL-Plotter/releases

lianzaozi, 2020-11-22:
Thanks for your hard work, I will do some tests.

I downloaded the source code of version 1.5.2.4, opened it with Visual Studio, compiled it and ran it, but the software showed that this was the source code of version 1.5.2.3.

svenhb, 2020-11-23:
As so often: I forgot to upload a files: 
- https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter/GRBL-Plotter.csproj

lianzaozi, 2020-11-23:
Thank you very much

I redownloaded the source code, and when it was compiled and run, it still prompted that the software version was 1.5.2.3.

svenhb, 2020-11-23:
Strange, if I open the file GRBL-Plotter.csproj with a text editor, the correct information is in "<ApplicationVersion>1.5.2.4</ApplicationVersion>".  
Perhaps close and open Visual Studio or reset the debugger.

lianzaozi, 2020-11-23:
The software version number was not changed, so I was prompted for version 1.5.2.3 when compiled and run
[image]

[image]

svenhb, 2020-11-23:
Ok, I updated this file also right now:
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter/Properties/AssemblyInfo.cs

lianzaozi, 2020-11-23:
Thank you for updating the file.

I have a question. In version 1.5.2.4, the arrows in the processing direction are displayed in the same direction, but this is not the case in the previous version. I think the arrow in the previous version is easier to understand.

Is there any special consideration in changing the software of version 1.5.2.4 to display the machining direction arrow direction in this way?

This is a screenshot of version 1.5.2.4 plotter software showing processing direction arrow
[image]

This is a screenshot of an earlier version of the plotter software showing the processing arrow
[image]

svenhb, 2020-11-23:
Looks like a new bug... I have to figure out.

lianzaozi, 2020-11-23:
I think the plotter software version 1.5.0.5 shows the processing direction and processing sequence Numbers in the correct way.

lianzaozi, 2020-11-24:
For the time being, I haven't found any problem with the function of switching machining directions. At present, The amount of testing I do is relatively small. Later, I will carry out more tests to verify again. Don't close it for a while

svenhb, 2021-07-31:
I did a little improovement: closed path can be rotated or reversed. Also I added a small direction arrow:
[image]

closed 2021-07-31 by svenhb