svenhb/GRBL-Plotter #140: Improve the left-click to select other geometric primitives

lianzaozi, 2020-08-05:
Thanks for your hard work

Now the plotter software has an inconvenient use, that is, when I select certain geometric primitives by left-clicking in the track display area, I often select other geometric primitives near this geometric primitive. But left-click on other geometric primitives can sometimes be accurately selected. This problem exists in every version of plotter software.

The version number displayed when the software of version V1.5.0.4 is opened is V1.5.0.3.

Test dxf file
[2007.zip](https://github.com/svenhb/GRBL-Plotter/files/5025601/2007.zip)

The circles and lines selected in the blue frame in the screenshot, I left-click on them, and the graphics near them are selected. When I zoomed in and displayed the track display area and left click on them, sometimes I could select them again.
[image]

The small cross mark in the gif animation that demonstrates the operation process is the position of my mouse. When I left click the mouse, the cross mark will have a highlighted yellow circle, by looking at whether there is a yellow circle at the cross mark Can tell if I pressed the mouse button.
[image]

svenhb, 2020-08-05:
Yes, I know. Up to know  I just look for known coordinates near by, which are known from G-Code. To find a line when clicking in the middle, I need to interpolate all possible coordinates inbetween...

lianzaozi, 2020-08-10:
Thank you for your reply.

svenhb, 2021-08-04:
Check new release for improovement: https://github.com/svenhb/GRBL-Plotter/releases

lslouka, 2021-08-05:
Hello.
I have a problem with compile firmware for ATMEGA 2560. When I connect atmega, I read only Missing 11 Real-time Status Reports per 10 seconds. 
If I use your hex file grbl-Mega-5X-v1.1l.20190605.hex all is working fine.
Can you please send me a GRBL code for arduino IDE from which was generate this hex file? I must set some settings for my machine.
My e-mail adress is: ace1x@centrum.cz

svenhb, 2021-08-06:
Hello,
as I remember I took the hex file directly from here: https://github.com/fra589/grbl-Mega-5X/releases.
But I'm not sure, so need to look on my computer.

lianzaozi, 2021-08-06:
When the software of V1.6.0.1 selects a straight line, other straight lines are also selected, and when an ellipse is selected, the elliptical arc will also be selected.

lianzaozi, 2021-08-06:
[image]
[image]
[image]
[2007_test_dxf.zip](https://github.com/svenhb/GRBL-Plotter/files/6946143/2007_test_dxf.zip)

svenhb, 2021-08-06:
Because the lines are placed in same figure (the two arcs also).
Will fix this

lianzaozi, 2021-08-16:
The software of V1.6.0.2 cannot open the dxf file. The dxf file "2007_test_dxf.zip" used in the test has been uploaded.
[image]

[image]

svenhb, 2021-08-17:
I updated the setup in the 1.6.0.2 release

lianzaozi, 2021-08-17:
What do I need to set up to read DXF files properly

svenhb, 2021-08-17:
>What do I need to set up to read DXF files properly 
 
What do you mean?

lianzaozi, 2021-08-17:
> > 我需要设置什么才能正确读取 DXF 文件
> 
> 你的意思是？

The latest version of the software can not load the DXF file, what can I do?

svenhb, 2021-08-17:
1.6.0.3 can : https://github.com/svenhb/GRBL-Plotter/releases/download/v1.6.0.2/GRBL-Plotter_1603_release.zip

lianzaozi, 2021-08-20:
hello

How to set slotting compensation in plotter software? If I cut the size of 100*100, in the laser cutting process, the laser will burn some of the material, resulting in the cut out of the size of less than 100*100

svenhb, 2021-08-20:
Try cutter radius compensation, use + or - radius for inner or outer contour:
Grey background path shows original path, before radius compensation. Result depends on figure path direction: clockwise or counter-clockwise.
[image]

lianzaozi, 2021-08-21:
Thank you for your answer. If I want to cut a circle, both inside and outside of it need to be compensated. Is there any way to do that?

lianzaozi, 2021-08-21:
Cut concentric rings

svenhb, 2021-08-21:
Perhaps, 
- load graphic, do radius compensation inside, cut inside
- load graphic, do radius compensation outside, cut outside 

I would create both cuts in LibreCad via Polyline - Equidistance polyline
[image]

lianzaozi, 2021-08-21:
Thank you for your answer. I don't quite understand. Does it have to be cut twice? Can it be solved once and for all? If there are too many concentric rings to cut, then the procedure is a little cumbersome

svenhb, 2021-08-21:
Perhaps I don't understand your needs: I though you want to cut a circle inside and outside the shape - for me the result is a ring.

lianzaozi, 2021-08-21:
yes

lianzaozi, 2021-08-21:
The effect of clicking the left button to select primitives has been improved a lot, which is very good. I will conduct more tests. The software version I use for testing is V1.6.0.5.

lianzaozi, 2021-08-21:
An error is reported when the software of the applicable version V1.6.0.5 opens the ".nc" file, which is automatically generated by the plotter software reading the dxf file.

nc file
[2007_test - 1_.zip](https://github.com/svenhb/GRBL-Plotter/files/7026316/2007_test.-.1_.zip)

dxf file

[2007_test - 1dxf.zip](https://github.com/svenhb/GRBL-Plotter/files/7026317/2007_test.-.1dxf.zip)

lianzaozi, 2021-08-21:
[image]

svenhb, 2021-08-22:
I updated the setup to 1.6.0.6
https://github.com/svenhb/GRBL-Plotter/releases/download/v1.6.0.5/GRBL-Plotter_Setup.exe

svenhb, 2021-08-22:
Also update of changed files: GCodeVisuAndTransform.cs, MainFormLoadFile.cs

svenhb, 2021-09-04:
I think it's solved and works

closed 2021-09-04 by svenhb