svenhb/GRBL-Plotter #141: Add an example to allow users to customize the generated GCode format

lianzaozi, 2020-08-07:
Wǒ jīntiān fāxiànle yīgè wèntí, wǒ yòng huìtú yí ruǎnjiàn bǎ dxf wénjiàn zhuǎn chéng GCode wénjiàn, ránhòu wǒ ràng yī tái jīguāng shèbèi dòu qǔ GCode bìng yùnxíng, dànshì shèbèi bàocuòle. Hòumiàn jīngguò wǒ de jiǎnchá, wǒ fāxiàn yǒu yīxíng GCode de hòumiàn méiyǒu kònggé, wǒ zhīdào huìtú yí ruǎnjiàn shēngchéng de GCode hòumiàn huì zìdòng jiā shàng kònggé. Zhège cuòwù kěnéng shì wǒ zài huìtú yí ruǎnjiàn wàibù shǒudòng xiūgǎi GCode bǎ xíng wěi de kònggé shānchúle yǐnqǐ de. Dànshì měi yīxíng hòumiàn yǒu méiyǒu kònggé, rén hěn nán qù pànduàn.

Jīyú yǐshàng kǎolǜ wǒ yǒu yǐxià de xiǎngfǎ
1. Suǒyǐ wǒ xiǎng huìtú yí ruǎnjiàn zài shēngchéng GCode de shíhòu, néng fǒu zuò gè TextBox kòngjiàn, jiǎrú wǒ zài zhège TextBox kòngjiàn zhōng tiánxiě fēn hào (;) zé shēngchéng de měi yīxíng GCode hòumiàn dōu yǒu yīgè fēn hào.(Zhège shèzhì xiàng zuò chéng kě xuǎnzé de)
2. Yǐ wǒ xiǎng huìtú yí ruǎnjiàn zài shēngchéng GCode de shíhòu, néng fǒu zuò gè TextBox kòngjiàn, jiǎrú wǒ zài zhège TextBox kòngjiàn zhōng tiánxiě"M10;G04 T50;"kěyǐ yǔnxǔ wǒ zài G00 hé G01/G02/G03 zhī jiān chārù"M10;G04 T50;".(Zhège shèzhì xiàng zuò chéng kě xuǎnzé de)
3. Yǐ wǒ xiǎng huìtú yí ruǎnjiàn zài shēngchéng GCode de shíhòu, néng fǒu zuò gè TextBox kòngjiàn, jiǎrú wǒ zài zhège TextBox kòngjiàn zhōng tiánxiě"M30;G04 T20;"kěyǐ yǔnxǔ wǒ zài G01/G02/G03 hé G00 zhī jiān chārù"M30;G04 T20;".(Zhège shèzhì xiàng zuò chéng kě xuǎnzé de)

zhèyàng jiù fāngbiàn yònghù shēngchéng fúhé zìjǐ de GCode géshì. Jù wǒ suǒ zhī xiànzài huìtú yí ruǎn kěyǐ yǔnxǔ yònghù zì dìngyì GCode de yè méi hé yè jiǎo, zhè fēicháng bàng.

Wèile shuōmíng wǒ de xiǎngfǎ wǒ zuòle liǎng gè tú lái shuōmíng,image1 shì yònghù méiyǒu shèzhì GGcode de géshì huìtú yí ruǎnjiàn shēngchéng de GCode yàngshì.Image2 shì yònghù shèzhìle yīxiē fúhé zìjǐ de tiáojiàn, huìtú yí ruǎnjiàn shēngchéng de GCode géshì.
展开
686/5000
I found a problem today. I used plotter software to convert the dxf file into a GCode file, and then I let a laser device read the GCode and run it, but the device reported an error. After my inspection, I found that there is no space after a line of GCode. I know that the GCode generated by the plotter software will automatically add a space after it. This error may be caused by manually modifying GCode outside the plotter software to delete the spaces at the end of the line. However, it is difficult for people to judge whether there are spaces after each line.

Based on the above considerations, I have the following ideas
1. So I wonder if the plotter software can be a TextBox control when generating GCode. If I fill in a semicolon (;) in this TextBox control, there will be a semicolon after each line of GCode generated. (This setting item is made optional)
2. I want to be able to make a TextBox control when the plotter software generates GCode. If I fill in "M10;G04 T50;" in this TextBox control, I can insert between G00 and G01/G02/G03 "M10;G04 T50;". (This setting item is made optional)
3. I want to be able to make a TextBox control when the plotter software generates GCode. If I fill in "M30;G04 T20;" in this TextBox control, I can insert it between G01/G02/G03 and G00 "M30;G04 T20;". (This setting item is made optional)

In this way, it is convenient for users to generate a GCode format that conforms to their own. As far as I know, the plotter software now allows users to customize the header and footer of GCode, which is great.

To illustrate my idea, I made two pictures to illustrate, image1 is the GCode style generated by the plotter software that the user has not set the GGcode format. image2 is the GCode format generated by the plotter software after the user sets some conditions that meet his own requirements.

image1
[image]

image2
[image]

svenhb, 2020-08-10:
For 2) and 3) the individual Pen up/down commands should work:
[image]

lianzaozi, 2020-08-11:
Thank you
For 2 and 3, I tested according to the method you said, and the test results fully meet my requirements

svenhb, 2020-08-15:
Now 1) is also implemented - I don't if it is useful. Perhaps indivdual line end disappears after G-Code transformation...
[image]

lianzaozi, 2020-08-16:
When the "apply text at line end" function is enabled, can the next version of the plotter software add the same symbol after each line of GCode? I have a good idea to call one when GCode is exported and saved A method to add symbols to each line of GCode. This may solve the problem that some CGode lines have symbols after them, and some CGode lines have no terminator after them, because in order to maximize the avoidance of other problems, you only need to traverse each line when saving (the lines with comments are ignored) GCode and then Add a symbol to the end of the line. The saved GCode will add a symbol at the end of each line.

svenhb, 2020-11-21:
Function is implemented

closed 2020-11-21 by svenhb