svenhb/GRBL-Plotter #146: Problems with opening dxf files in plotter software V1.5.0.0 or later

lianzaozi, 2020-08-10:
There is a problem with the plotter software V1.5.0.0 and later versions. There will be problems when opening the 1.dxf file, but it can be opened normally with the software version before V1.3.4.6

Test dxf file
[1.zip](https://github.com/svenhb/GRBL-Plotter/files/5049930/1.zip)

svenhb, 2020-08-10:
LibreCad shows also errors when loading...
I need to figure out how to handle this problem
[image]

lianzaozi, 2020-08-10:
Thank you for your reply. This is a screenshot of the "1.zip" file opened with the software version V1.3.4.6.
[image]

svenhb, 2020-08-15:
Should be fixed. Reason: somwhere is anArc with r = 1.1e-13 too small for float, caused error when drawing an Arc.

lianzaozi, 2020-08-18:
Thank you for fixing the problem in time, and I will conduct some tests.

svenhb, 2020-10-17:
I hope your tests were succesful

closed 2020-10-17 by svenhb