svenhb/GRBL-Plotter #147: Open svg file software does not respond for a long time

lianzaozi, 2020-08-10:
When opening the SVG_Test1.zip file, only the use case "example_laser.ini" can be opened normally. However, when other use cases are selected, it cannot be opened normally, and the software displays no response for a long time. When you open the SVG_Test1.zip file with a version before V1.3.4.6, you can select any use case software to work normally

Test svg file

The SVG_Test1.zip file for testing was uploaded when reporting issue #138. Please use this file to test(https://github.com/svenhb/GRBL-Plotter/issues/138)

svenhb, 2020-08-10:
By default "Grouping" and "Sort objects" are enabled. Your example file is very big (25147 elements) and sorting and reversing paths takes a long time...
Perhaps I should estimate object-count by file size and automaticly disable the group option?

lianzaozi, 2020-08-10:
Why is it that when I choose the "example_laser.ini" use case, it opens quickly. Is it because the "example_laser.ini" use case does not enable the functions of "grouping" and "sorting objects"?

When opening the file, a pop-up window will prompt the user when the collection graphics exceed 25147 elements. If the "grouping" and "sorting objects" functions are activated, it will take a lot of time. Whether to disable the "grouping" and "sorting objects" functions, if the user clicks "OK button" then the software opens the file by disabling the "grouping" and "sorting objects" functions. If the user clicks the "cancel" button, it opens the file by enabling the "grouping" and "sorting objects" functions.

lianzaozi, 2020-08-10:
All the primitives in SVG_Test1.zip are spline curves. The entire svg file is composed of 503 spline curves.

svenhb, 2020-08-10:
[image]

svenhb, 2020-08-10:
>The entire svg file is composed of 503 spline curves.

Which results to 499 figures, containing 50 G1 moves in average... 
[image]

lianzaozi, 2020-08-11:
thank you for your reply.
You are right, one spline curve is transformed into many small straight lines, so the total number exceeds 25147.

svenhb, 2020-08-15:
Should be fixed now. Grouping will be discart if only one property-value is found:

lianzaozi, 2020-08-16:
Thanks for your hard work

I don’t quite understand the meaning of "Grouping will be discart if only one property-value is found"?

Now the software can open the SVG file within the acceptable time. I will conduct some additional tests.

svenhb, 2020-08-16:
For example you group by colors, then you would get a group for "red" and "blue" and "green".
If there is just only one color (e.g. "red"), grouping makes no sense and no grouping will be performed.

lianzaozi, 2020-08-18:
Thank you for your reply.

closed 2020-10-17 by svenhb