svenhb/GRBL-Plotter #148: Chinese translation error correction

lianzaozi, 2020-08-11:
Copy the Chinese in the document and replace the Chinese in the original location.
[Chinese translation error correction.zip](https://github.com/svenhb/GRBL-Plotter/files/5055635/Chinese.translation.error.correction.zip)

lianzaozi, 2020-08-12:
supplement
decimal places=======>小数位

svenhb, 2020-08-15:
I took your translations, thanks.

lianzaozi, 2020-08-17:
Thank you for correcting some Chinese translations, the Chinese translation in the screenshot below also needs to be corrected

MainForm
[image]
Ruler==========>标尺

[image]
Pen-Up path==============>笔上升路径(快速定位)

ControlSetupForm
[image]
Min===================>最小

svenhb, 2020-10-17:
Thank for your help

closed 2020-10-17 by svenhb