svenhb/GRBL-Plotter #149: Using Multiple Fiducials for modifying the cutting file

futurau, 2020-08-13:
Hi sven

I have a A3 printer and I can print on fabrics. I want to cut this printed fabrics. But the printed artwork deforms some time 2-3mm.

Feature request:
I want to put some registration marks (fiducials) and read the marks with a camera to correct the cut path automatically.

Is it possible to add this capability? 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

svenhb, 2020-08-13:
Hello,
it is already implemented, but I haven't tried for a long time...   
https://github.com/svenhb/GRBL-Plotter/wiki/PCB-drilling 
Let me know if it works, or how to improove.

futurau, 2020-08-13:
Thank You
I will try asap.

svenhb, 2020-11-21:
Hope it's working

closed 2020-11-21 by svenhb