svenhb/GRBL-Plotter #151: Surface scan - not working

AlexTimofte, 2020-08-20:
Hello Svenhb,

I'm trying to get a correct height map, but without results.
For example I placed a coin on a PCB (the coin has a thickness of 2 mm) I did the settings (area, depth, save height, speed) and I start to generate height map from surface scan. The result is a wrong map, I think that is not about offset or cut off. 
Maybe I skipped something in setup. The GRBL-Plotter version that I'm using is version 1.5.0.5.
Could you help me? 

[image]
[image]

Best regards,
Alexandru

svenhb, 2020-08-20:
What is wrong with the map? What do you expect?
For me, the map looks right, but the offset is perhaps wrong - I assume you don't want that Z-depth.
I didn't try this option for a long time, I need to check by myself how to use it.
Did you check this manual: https://github.com/svenhb/GRBL-Plotter/wiki/surface-scan ?

I woud go to this position (because it shows a depth of 0) and zero the z axis on surface, then apply height-map (with max. value set to 0.0) to the G-Code...
[image]

AlexTimofte, 2020-08-21:
Hello Svenhb,

Thank you for your answer.
I was expecting to see a clear contour, even if there will be an offset.
Another thing is that, if I scan the surface several times, in the same conditions (position, area, settings, starting from 0 Z), each time I get a completely different map comparing with previous one.
[image]

In "Probing" view I observed that the z position was read right, but in surface scan windows, the color and position seems to be not right placed.  
[image]

I will double check the manual. 

Regards, 
Alexandru

svenhb, 2020-08-21:
To see the relief of the coin you need a smaller raster (grid-size), in my example I used 0.2 mm. Your first screenshot shows 5 mm.  

The message in the main window shows the result of grbl in machine coordinates. But the scan window recalculates it to work coordinates.

Without pressing "set max. value = 0.0" the result picture should show colors from minimum depth to maximum depth - usually not starting at 0.000. 

Perhaps the old scan was still in the memory and not cleared - overlapping (with min. max. values) the new scan?

svenhb, 2020-08-21:
More information here: http://svenhb.bplaced.net/?CNC___GRBL-Plotter___Hoehenprofil
and here: https://www.youtube.com/watch?v=KZMYFQ_tFOc&t=1s

AlexTimofte, 2020-08-24:
Hi Svenhb,

For sure, I'm agree with you, for a fine contour, the scanning must be finer, my complaint is related to resulted height in the height map.
The resulted map does not reflect the reality (even if will be shifted or re scaled).
My example can be seen as a worst case probe where I have a kind of step in the work plan.
I also tried with an inclined plan, but the results were not as expected.
[image]

svenhb, 2020-08-24:
But probing works like this: starting from zero it is moving down to negative values until contact.   
The delta shows the thickness of your coin (-1.9 - (-3.9)) = 2
[image]

AlexTimofte, 2020-08-24:
Ok, this aspect is already understood. and that's what I expect to happen.

[image]

svenhb, 2020-08-24:
Did you watch the surface scan, when scanning the coin? Perhaps you scanned on wrong coordinates - without coin.   
You need to zero X and Y at the lower-left edge of the scanning area...
I tried the surface scan at the weekend, and it was ok.

closed 2021-05-02 by svenhb