svenhb/GRBL-Plotter #153: Loosing settings when copy-pasting the solution

amoineau, 2020-10-16:
**Describe the bug**
I'm using your amazing work to build an HMI for a machine. Among other modifications, I wanted to change the custom buttons to fit my needs. I tried changing through the Settings in VS but it wasn't updated when launching the app. I then realised that I could change them at runtime through the setup menu. This worked fine but when I tried to copy-paste the solution (or even the build) to an USB stick to bring it to the PC connected to the machine I realised that everything was back to default.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the newest release
2. Open the app and change a custom button behavior
3. Close eveything
4. Copy paste the solution somewhere else on your drive
5. Open the copy

**Expected behavior**
The changes should be visible on the copied version.

Thank you very much for your help and for all the hard work you must have put in this project !

svenhb, 2020-10-16:
Thanks for the detailed description. You have two options:
1) Export and import the setting to an INI file:
[image]

2) Or find the location of the properties (somewhere in c:\User\YourUser\AppData\Local\GRBL_Plotter\someStrangePathLike GRBL-Plotter.exe_Url_dfcqomyqns4mxblxdl1jbpyubn5ak31o\Version\user.config.
Copy this user.config and paste it to your new location....

svenhb, 2020-10-17:
As shown here https://grbl-plotter.de/index.php?id=auto-pen-change you can now drop INI file to change settings.

closed 2020-10-17 by svenhb

amoineau, 2020-10-19:
Thanks for the swift reply ! I feel kind of dumb... A week ago I wondered what this machine config meant and when I reached my problem I didn't put two and two together.