svenhb/GRBL-Plotter #154: The software of version 1.5.2.0 has some problems in the function of x=0 and y=0 when importing DXF graphics.

lianzaozi, 2020-10-19:
When using version 1.5.2.0 software to import dxf graphics, check the option "Offset graphics origin to 0;0". The dxf graphic is not offset to the coordinates x=0, y=0

In the 1.5.0.5 version of the software, the offset is 0; the 0 function can work well.

The screenshot image1 below is a screenshot of using 1.5.0.5 version software to test the offset to coordinates x=0 and y=0. The screenshot image2 is a screenshot that uses the 1.5.2.0 version of the software to test the offset to the coordinates x=0 and y=0.

image1  （using 1.5.0.5 version software）
[image]

image2 （using 1.5.2.0 version software）
[image]

svenhb, 2020-10-22:
Check new version https://github.com/svenhb/GRBL-Plotter/releases/latest

lianzaozi, 2020-10-28:
Thank you, the problem is solved

svenhb, 2020-11-21:
Should still work

closed 2020-11-21 by svenhb